slic3r/Slic3r #4103: GPL violation? XYZware

magnusjjj, 2017-08-31:
Hi. After poking around XYZWare, i found its slicer, and extracted it. They statically linked with perl, which is its own issue, but in it was some code that looked... rather like slic3r.

You can see it at:

https://www.diffchecker.com/Cxcyu4Tm

which is the xyzware version vs https://github.com/alexrj/Slic3r/blame/37105e8237272565b60550c41945b3c41b11a9d9/lib/Slic3r/Print/Object.pm

Its a four year old version they are using, only lightly hacked.

lordofhyphens, 2017-08-31:
I can't say I'm surprised in the least, @alexrj needs to weigh in. 

That the comments are the same too is pretty telling.

magnusjjj, 2017-08-31:
They are also using several other open source projects, with a range from simple libraries, some gpl, lgpl, mit, to repetier where they just removed nearly all the functionality + donate button and any mention.. which is legal according to the license, but still, dick move. This is a fair bit plaincut and not so much edgecase :).

Need a dump?

magnusjjj, 2017-08-31:
User License Agreement

This user license agreement (this "Agreement") governs your use of XYZWare software ("XYZWare") and defines the relationship between you, or in the case that you represent and are using XYZWare on behalf of a company or other entity, that company or other entity ("you") and XYZprinting, Inc., a Taiwanese company ("XYZprinting"). 

PLEASE READ THIS AGREEMENT BEFORE PRESSING THE "AGREE" BUTTON AND CHECKING THE BOX WHERE THE OPTION IS MADE AVAILABLE TO YOU. BY PRESSING "AGREE," YOU REPRESENT AND WARRANT THAT YOU HAVE THE AUTHORITY TO ACCEPT THIS AGREEMENT AND YOU ARE AGREEING TO BE BOUND BY THE TERMS OF THIS AGREEMENT. IF YOU DO NOT AGREE TO THE TERMS OF THIS AGREEMENT, PRESS "CANCEL" AND STOP USING AND/OR DOWNLOADING XYZWare. 

1.    XYZWare
XYZWare means, collectively: 
(i)  all the contents of the disk(s), CD-ROM(s), electronic mail and its file attachments, or other media with which this Agreement is provided, including the object code form of the software delivered via a CD-ROM, electronic mail or web page;
(ii)  digital images, stock photographs, clip art, or other artistic work; 
(iii)  related explanatory written materials and any other possible documentation related thereto; 
(iv)  upgrades, modified versions, updates, additions licensed to you by XYZprinting. 

2.    LICENSE
2.1  XYZWare may be provided separately or in the form of pre-installed, embedded software on a XYZprinting product.. By downloading or using the XYZWare you agree to be bound by the terms of this Agreement.  If you do not agree to the terms and conditions of this Agreement or do not have the authority warranted above, do not download or use the XYZWare.
2.2  Subject to the terms and conditions of this Agreement, XYZprinting grants you a nontransferable, nonexclusive, royalty-free, fully paid, revocable, worldwide license, without the right to sublicense, to permit individuals authorized by you ("Authorized User") to install, execute and display XYZWare, in executable object code format only. 

3.    RESTRICTIONS  
3.1  The rights granted to you under this Agreement are subject to the following restrictions: 
(a)  you shall not license, sell, rent, lease, transfer, assign, distribute, host, outsource, disclose or commercially exploit XYZWare or make XYZWare available to any third parties other than Authorized User;
(b)  you shall not modify, make derivative works of, disassemble, enhance or merge the whole or any part of XYZWare with any other software, source code or documentation;
(c)  you shall not reverse compile or reverse engineer any part of  XYZWare, except and to the extent that the applicable law expressly permits doing so;
(d)  you shall not access XYZWare in order to build a similar or competitive product or service; 
(e)  except as expressly stated herein, no part of  XYZWare may be copied, reproduced, distributed, republished, downloaded, displayed, posted or transmitted in any form or by any means, including but not limited to electronic, mechanical, photocopying, recording or other means; 
(f)  any future release, update, or other addition to functionality of XYZWare will be subject to the terms of this Agreement, unless XYZprinting expressly states otherwise; and
(g)  you shall only use XYZWare in a manner that complies with all applicable laws in the jurisdiction where you use XYZWare, including but not limited to applicable restrictions regarding copyright and other intellectual property rights. 
3.2  Neither XYZprinting nor any of its suppliers is obligated to provide any services, updates or upgrades to  XYZWare.  You shall not remove, obscure or alter any copyright and other proprietary rights notices in XYZWare or any  copies thereof.

4.    OWNERSHIP  
All right, title, and interest, including all intellectual property rights, in and to  XYZWare shall be owned and retained by XYZprinting and its licensors.  Any rights not expressly granted by XYZprinting under this Agreement are reserved, and except as expressly granted in this Agreement, XYZprinting grants no rights or licenses to you, by implication, estoppel, or otherwise in or to XYZWare.  You acknowledge that you acquire no ownership interest in XYZWare by downloading or using XYZWare.

5.    Collection of Non-Personal Information
5.1 You acknowledge and agree that XYZprinting will collect data and information in a form that does not create direct association with any specific individual by itself ("Non-Personal Information). XYZprinting may collect, use, transfer, and disclose non-personal information for any purpose.
5.2 Non-Personal Information includes but not limited to series number of the 3D printer and the time you use XYZWare and the 3D printer. 

6.    DISCLAIMER OF WARRANTIES  
6.1  You shall not use XYZWare in any situations which health or safety might be at risk.  You will not use XYZWare in connection with the operation of nuclear facilities, aircraft navigation, communication systems, medical devices, air traffic control devices, or other situations in which the failure of XYZWare could foreseeably lead to death, personal injury, or physical property.
6.2  XYZprinting makes no representations regarding fitness of XYZWare for the design, construction of parts or components that may be subject to additional regulation in your jurisdiction, such as toys, weapons or medical devices.
[... cut ...]
13.6  This Agreement will be interpreted without application of any strict construction in favor of or against you or XYZprinting.  
13.7  You agree that any notice may be provided for you by email.
13.8  You may not assign, delegate or transfer, including by operation of law, sale of assets, merger or otherwise, this Agreement or any of your rights or obligations (in whole or in part) under this Agreement to any third party.

QUESTIONS OR ADDITIONAL INFORMATION.  
If you have questions regarding this Agreement, or wish to obtain additional information, please send an e-mail to our Customer Service: support@xyzprinting.com.

lordofhyphens, 2017-08-31:
Well anything that is MIT/BSD is legit to do what they want. But GPLv3 (which is what Slic3r is under) requires that source modifications (which hey, they made) are available.

bubnikv, 2017-08-31:
Interestingly enough, the new Simplify3D produces a "new" gap fill with the
familiar "trimmed skeleton" pattern. From how it works, it looks like they
used the Slic3r gap fill code verbatim.

lordofhyphens, 2017-09-02:
@magnusjjj is that the old open-licensed repetier they jacked or the newer closed source repetier?

magnusjjj, 2017-09-03:
@lordofhyphens : What licenses has the project had? I linked the revison for that *one file* that is the closest I could find energy for that day. Wouldn't know any closer to that.

magnusjjj, 2017-09-03:
The revision itself is something like 4 years old.

lordofhyphens, 2017-09-03:
As far as I know Slic3r has always been GPL.

I was wondering about RepetierHost  (which you has mentioned that they had
used it whole with the branding ripped out).

magnusjjj, 2017-09-03:
For that, I have zero clue. The folks over at RepetierHost seemed to think it was fine (I opened a ticket there, but they have the apache license). I have just been decompiling it half arsedly, trying to decipher things like how the firmware is uploaded and some drm checks. That, however, does not translate well to comparing with a git repo :).

bubnikv, 2017-10-14:
I have reported the GPL violation of Slic3r by XYZprinting to the free software foundation and I have asked them to take a legal action.
license-violation@gpl-violations.org

alexrj, 2018-01-04:
Relevant: https://www.cc.gatech.edu/~rduan9/publications/osspolice.pdf