slic3r/Slic3r #4106: FIRMWARE RETRACT G10 S1 BEFORE TOOL CHANGE PLEASE

studiodyne, 2017-09-02:
**## No G10 S1 before ToolChange**

Firmware retraction when ' selected ' , forget to make ' tool change retractation 
Please implement  G10 S1 just before a toolchange , to make the long retractation , if a dual nozzle the print is still possible but corrupted but if SINGLENOZZLE ' Y ' , with 50mm load/unload the print crashes !

Easy to code , just REPLACE the G10 by a G10 S1 before tool change 

I have made a topic about ' Firmware retractation ' here [Understanding FIRMWARE RETRACTATION G10 / G11 / G10 S1 #7597](https://github.com/MarlinFirmware/Marlin/issues/7597)

A piece of code of slic3r
```

G1 X176.483 Y221.677 F7800.000
G1 X178.271 Y223.465 E924.73334 F1800.000
G10 ; retract    ************* NO SWAPPING RETRACTION AND NEED A G10 S1******
G92 E0
T1 ******* NOT RECOVERED OR LOADED OR PRIMED *********
G92 E0
G1 X200.222 Y222.931 F7800.000
G1 X200.154 Y223.286 E0.09169 F1800.000
G1 X199.990 Y223.777 E0.22301
G1 X199.819 Y224.121 E0.32070
```

lordofhyphens, 2017-09-02:
There's a pre-toolchange custom gcode (as well as a post-toolchange custom gcode).

studiodyne, 2017-09-02:
If i use your ‘ script tab’ , and place a G10 S1 on the tab , on the produced Gcode i will have
G10 ; retract     ‘ the normal retract ‘
G10 S1              ‘ my added gcode ‘
It’s not serious , to confuse the machine with 2 instructions , and tweaking marlin just for a bad gcode production is not serious too
For the moment , SLIC3R not ready and not scriptable for Multi extruder and Firmware retractation

But it's possible to make this , and after production of gcode , to open it , in a text editor and replace 
G10 ; retract     ‘ the normal retract ‘
G10 S1              ‘ my added gcode ‘
By only this 
G10 S1              ‘ my added gcode ‘

@lordofhyphens ' if you read better what i say , you'll never post this comment' i have already talk about this problem of scripting

lordofhyphens, 2017-09-02:
Okay, let's go back to the beginning here so I can better understand the problem.

@studiodyne, I need you to provide the version of Slic3r you are using (if it's not 1.3.0-dev from https://dl.slic3r.org/dev then you need to confirm that it is still a problem there) as well as an exported configuration (use Ctrl-E). This information was originally requested in the text you deleted to make your post.

You can zip the config and drop it here.

studiodyne, 2017-09-02:
I use 1.29 , but  i decide to follow you and i go to dwl the 1.3 , i will make my experience and i will come back here to give my results
Now i have to go out , but in a few hour i will work on it 
Thank you very much for your help
 👍

studiodyne, 2017-09-02:
Look i have remade the first comment for more comprehensive explanation

studiodyne, 2017-09-02:
I use 1.3 Dev , and it's the same problem 
`TOOLCHANGE RETRACTION not implemented 👎 `

Hope devs will make a fix in the 1.3 release 

Steeve

lordofhyphens, 2017-09-02:
Hint: I am one of the devs. Please do [what I asked you to do](https://github.com/alexrj/Slic3r/issues/4106#issuecomment-326744360).

studiodyne, 2017-09-02:
Ok i will explain ...  With no firmware retraction activated , all works ...  
But with firmware retract enabled , in only one extruder , all works
But in multi extruders , printing is impossible with firmware retractation activated
Why .? 
Because SLIC3R forget to make the ' swap retractation ' that is a long retraction , and just make a G10 ' a short retractation '
The process must be like this

T0 is used to print  all i ok 
G10 S1   ' long retractation  50mm for SINGLE NOZZLE SYSTEM  10mm for dual nozzle system '
T1     tool change ' that is already retracted at 50mm or 10mm 
G11   recovery  and loading '  50mm of loading or 10mm of loading 

and print continue

But with slicer   in multi extruder mode we have
T0    printing 
G10   short retract
and no G11 to recover , all is buggy

READ slowly my first comment on the top of the page and you will understand all

---------------------------------------------------------------------------------------------------
For the moment , during the time you make a fix , i think i can use the script processor , to make a good gcode file 
I'm working on it 

Please can you send me a link of the   SCRIPT functionnality , to help me to write my script

Thks 👍

studiodyne, 2017-09-02:
You told me to send you , the configuration , but i have no config ,for the moment , i make 2 extruder machine setting  , and i make a dual printing , and i see the gcode
Settings of nozlle and others , will not interest me if the engine don't produce correct gcode

studiodyne, 2017-09-02:
I don't have '  before tool change ' and ' after tool change ' tab
Please explain me how i can have it 
Thks

lordofhyphens, 2017-09-02:
[image]

Use that option to export a copy of your Slic3r config that is showing the problem. Then add it to a zip file and in the Github issue interface, attach the zip file (email attachments DO NOT WORK).

[image]

studiodyne, 2017-09-02:
[Slic3r_config_bundle.zip](https://github.com/alexrj/Slic3r/files/1272236/Slic3r_config_bundle.zip)

studiodyne, 2017-09-02:
If you find me the ' BEFORE TOOL CHANGE ' + ' AFTER TOOL CHANGE ' 

PROBLEM IS SOLVED , but i don't find it

studiodyne, 2017-09-02:
/play

studiodyne, 2017-09-02:
@lordofhyphens 
*  _I have a problem , may be i'm mad ... two hours ago I told you that i have seen the ' tabs post tool change , and after tool change ' , now I'm back at my home and tabs have disappeared ... I'm mad or the sofware change the tabs . Now i have downloaded dev version and Tabs not here , always the single toolchange tab .............._

------------------------------------------------------------------------------------------------
**Job for you , please make me a script , because I don't know Perl language**
------------------------------------------------------------------------------------------------

studiodyne, 2017-09-02:
@lordofhyphens 
*  _I have a problem , may be i'm mad ... two hours ago I told you that i have seen the ' tabs post tool change , and after tool change ' , now I'm back at my home and tabs have disappeared ... I'm mad or the sofware change the tabs . Now i have downloaded dev version and Tabs not here , always the single toolchange tab .............. it's not important i just need the ' before tool change '_

------------------------------------------------------------------------------------------------
**Please make me a script , because I don't know Perl language**

You must ' REPLACE' this block of code 

```
G10 ; retract
G92 E0
DELETE
```
By This 
```
G10 S1 ; Swapping Retract
```
And exactly the same with ' T1 '

_' DELETE ' is a word added in the toolchange TAB to detect the block that must be scripted_

lordofhyphens, 2017-09-02:
#4108 Hopefully it doesn't break anything else.

studiodyne, 2017-09-02:
No firmware retract ---> right gcode production
Activated firmware retract ---->  Not perfect cloning of the ' no firmware retract gcode production' and forgotten instructions

You can trust , because , you make what already works fine without it
The job is to simply make a perfect cloning without the firmware retract instructions

Thks for the PR 👍

lordofhyphens, 2017-09-02:
The PR issues G10 S1 if and only if FW retract is enabled and your firmware flavor is RepRap or Repetier.

That's all it does.
https://dl.slic3r.org/dev/win/branches/Slic3r-long-retract.2017.09.02.1338.c80de49.64bit.zip for a build.

studiodyne, 2017-09-02:
For Marlin code it's perfect , we have rewritten the firmware retract a few days ago
For information , i think marlin have the better retractation system , and may be in a few days if pull request of my suggestion accepted , marlin will have the better system with full option as possible
Zlift feedrate by lcd and gcode indepently of print feedrate
Long retract lengh and recover by gcode ' already implemented'
Long retract and recover feedrate by gcode too
and waiting an soon PR to have zlift feedrate by lcd and gcode 
M207 and M208 overloaded of arguments !

Thk you very much 
Retract indepently of the extrusion multiplier
Gcode total tweaking , with   zlift feedrate ' soon ' , long recover feedrate ' implemented ' 
and all is possible

lordofhyphens, 2017-09-02:
We mostly just deal with/react to the things that the firmware authors do. I don't use Marlin at all on any of my printers, so these syntax changes are news to me.

studiodyne, 2017-09-02:
It's why the code not really finished , if you use marlin , I'm sure this PR was already made :smile 👍 

Marlin is great and have features i found anywhere else , and I don't need 32 bits power , because delta printer cinematic don't please me too
What is your Soft and Your Hard and may be if you  have a  commercial printer instead of a custom made machine ?
My printer is a chinese 400x400 , my extruder is a single nozzle with direct drive ' made with cnc by myself' , the fans are a cutted ultimaker holder :smile:  , my board is a MKSGEN 1.4 with minipanel
And marlin give me features i love
UBL ---- > autoleveling = PERFECTION !!!!!!
Firmware retract  the most advanced !
🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 

> And the best of MARLIN is that  IT WORKS PERFECT , print always finish without an OS crash !

🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹 🔹

studiodyne, 2017-09-02:
@lordofhyphens 
Please can you add a new correction in this PR 

A G10 after a tool change is uneeded , because already retracted , and can corrupt ' the swap status of the machine for this extruder 

Look 
```
G1 X78.792 Y64.434 E84.08525
G10 S1   ********* ok ************
G92 E0   ********* uneeded but not dramatic ************
T1           ********* ok ************
G92 E0   ********* uneeded but not dramatic ************
G10        ********* uneeded and DANGEROUS FOR THE MARLIN   ************
G92 E0   ********* uneeded but not dramatic ************
G1 X102.896 Y111.253 F7800.000
G11 ; unretract
G92 E0
G1 F1800
G1 X102.856 Y111.848 E0.01842
```

lordofhyphens, 2017-09-02:
Two of my printers were made by me and use smoothieboards. So no marlin. I had AVR-based electronics at one point but did not care for it. 

The order of operation should be:
````
Retract
Tool Change
Unretract. 
````

studiodyne, 2017-09-02:
Yes Perfect !

:fr: :fr :fr: Smoothie :fr::fr::fr: 
😃

lordofhyphens, 2017-09-02:
See Gcode.cpp line 720 and then GCodeWriter.cpp lines 455, 466-503 if you want to see the order of operations. 

As a note: Tool Change custom gcode (there's only one) occurs prior to the toolchange command, but after the retract.

studiodyne, 2017-09-02:
Yes , you have a beautifull source code , but , you know this code is not a code i work every day and if will work on it a few hour , i will have a better reading loll but for the moment , it's hyeroglyph , and i understand nothing , and impossible to see an ' order ' in this jungle lollll 
I know marlin code it's enough  😅

studiodyne, 2017-09-02:
Slic3r-long-retract.2017.09.02.1341.78a8f92.64bit
tested but no result , always the G10 after toolchange and some G92 E0 uneeded
😢

lordofhyphens, 2017-09-02:
https://ci.appveyor.com/project/lordofhyphens/slic3r/build/job/c73c3xbwxnpgwkq9

lordofhyphens, 2017-09-02:
https://github.com/alexrj/Slic3r/pull/4108/files#diff-ab51cc0eabff6301965ef4ed1f6357d5 Those are the tests that get executed on build, and they're passing now. 

The G92 E0 is always going to happen, that's part and parcel of how Slic3r maths out the next section.

studiodyne, 2017-09-02:
i have tried the last release 23.06 on the dev branch 

always this ' G10 ' after tool change , and some uneeded G92 E0

studiodyne, 2017-09-02:
ok i'm waiting

studiodyne, 2017-09-02:
A little issue again
The first tool change of  each extruder have a ' G10 ' after 
Only the first T0 and the first T1

If you solve it , job is perfect !!!!!!!!!!!!
😃

lordofhyphens, 2017-09-02:
Also remember to set "Retraction when tool is disabled" to 0.

studiodyne, 2017-09-02:
```
M109 S200 T0 ; set temperature and wait for it to be reached
M109 S200 T1 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
T0
G92 E0
G1 Z0.350 F7800.000
G10
G92 E0
G1 X69.551 Y61.320 F7800.000
G11 ; unretract
G92 E0
G1 F1800
G1 X70.986 Y59.736 E0.06612

G1 X78.792 Y64.434 E84.08525
G10 S1
G92 E0
******************************************************************************************************************************************************************
T1
G92 E0
G10
G92 E0
G1 X102.896 Y111.253 F7800.000
G11 ; unretract
G92 E0
G1 F1800
G1 X102.856 Y111.848 E0.01842
G1 X102.757 Y112.347 E0.03419
G1 X95.986 Y114.239 E7.74257
G10 S1
G92 E0
******************************************************************************************************************************************************************
T0
G92 E0
G1 X102.645 Y96.113 F7800.000
G11 ; unretract
G92 E0
G1 F3600
G1 X101.501 Y96.204 E0.03189

G10 S1
G92 E0
******************************************************************************************************************************************************************
T1
G92 E0
G1 X101.274 Y114.853 F7800.000
G11 ; unretract
G92 E0
G1 F900
G1 X100.751 Y115.201 E0.01745
G1 X100.217 Y115.461 E0.03392
G1 X99.674 Y115.644 E0.04985
```

studiodyne, 2017-09-02:
ok i try it

studiodyne, 2017-09-02:
where is this setting i don't find it ???
Retraction when tool is disabled" to 0.???

studiodyne, 2017-09-02:
this ' 0 ' values makes 2 advertising POP UP WINDOWS in the gcode generation
Are you sure this value must be setted to zero ?

studiodyne, 2017-09-02:
and always  the ' G10 ' after  the first tool change 

I think this ' G10 ' is an initialisation and appears only one time

Can you solve this ? If solved , job is finished 
👍

studiodyne, 2017-09-02:
> _note  , when we will have finished this PR ,  you can ' if you have time to waste ' , apply ' desactivation' on the ' retract setting tab ' , of the retract lengh and feedrate of the ' tool change retract edit box '_

lordofhyphens, 2017-09-02:
You're conflating two different retracts, I think. 

Load this config.ini and take a look; I've marked each toolchange with some custom gcode. Note the retract before and after it. 
[multi-extruder-test.zip](https://github.com/alexrj/Slic3r/files/1272564/multi-extruder-test.zip)

lordofhyphens, 2017-09-02:
There, GUI should disallow you from setting troublesome options now.

studiodyne, 2017-09-02:
hey , your config works
I just changed ' marlin ' , and disabled ' volumetric

studiodyne, 2017-09-02:
i have a problem ,  if i load your config  ,   G10 appears before tool change

lordofhyphens, 2017-09-02:
When the new build is done, use that.

studiodyne, 2017-09-02:
Another problem   ZLIFT , is not deactivated  when Firmware Retractation enabled
and in the gcode we have zlifting  mixed with g10 and g11

First , zlift must be disabled , to clean the gcode and may be some intermediate G10

lordofhyphens, 2017-09-02:
Set it to 0 in your configuration then. There's some use cases for keeping it available during FW retract.

studiodyne, 2017-09-02:
For the next PR , you must disabled ' ZLIFTING ' to avoid mixing of zlift and firmware retract
The marlin FW , have zlift included

studiodyne, 2017-09-02:
ok

lordofhyphens, 2017-09-02:
There's also no particularly bad side effects to having additional z lift from the slicer. Having the extra retract/restore was completely unnecessary and caused some bad side effects (which is why I put the guards in against it).

studiodyne, 2017-09-02:
```
You're conflating two different retracts, I think.

Load this config.ini and take a look; I've marked each toolchange with some custom gcode. Note the retract before and after it.
multi-extruder-test.zip
```
i didn't see some ' custom label or anything that labellize your gcode anywhere '
look
```
1 X127.172 Y130.370 E16.83404 ; skirt
G1 X127.370 Y130.099 E16.88800 ; skirt
G1 X127.605 Y129.844 E16.94374 ; skirt
G10 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.800 F7800.000 ; lift Z
G1 X141.613 Y133.330 F7800.000 ; move to first perimeter point
G1 Z0.300 F7800.000 ; restore layer Z
G11 ; unretract
G92 E0 ; reset extrusion distance
G1 F1800
G1 X140.974 Y133.758 E0.12351 ; perimeter
G1 X140.202 Y133.914 E0.25007 ; perimeter
G1 X139.426 Y133.755 E0.37737 ; perimeter
```

studiodyne, 2017-09-02:
Can you explain me why , if i use your config printer , i have the G10 before tool change , and if use my own quick setting , G10 is not before the tool change
How it's possible  ?

lordofhyphens, 2017-09-02:
Your own settings are likely incorrect in some way.

studiodyne, 2017-09-02:
my setting make a good gcode , your setting make a bug 
and it's my settings are wrong lol
I will try to find what is the reason of this g10 appears

lordofhyphens, 2017-09-02:
As a reminder, https://github.com/alexrj/Slic3r/pull/4108 has the current commit status of the branch.

studiodyne, 2017-09-02:
The last release work with my settings
Marlin + 2 extruders + firmware  and nothing else
But YOUR CONFIG generate a G10 before tool change 
I'm searching what make this

studiodyne, 2017-09-02:
IF volumetric is enabled  G10 APPEARS
Include ' volumetric mode ' in the same flavor ' if possible '

studiodyne, 2017-09-02:
no it's wrong , it's not the volumetric , please wait i'm trying again

studiodyne, 2017-09-03:
I've found the issue
It's ' FORCE RETRACTION LAYER CHANGE' 
I'm ok to leave this functionnality , but , is it possible to disable it if just after a layer change appears a tool change 
Can you make this conditionnal ?

lordofhyphens, 2017-09-03:
It already is conditional. Uncheck it to turn it off.

[63 more replies]

closed 2018-01-11 by lordofhyphens