slic3r/Slic3r #4111: Launch GUI by default

probonopd, 2017-09-02:
### Version

Slic3r-1.3.0-dev-66074f5-x86_64.AppImage

### Operating system type + version

Ubuntu 14.04

### Behavior
* Double clicking the AppImage or launching it from the menu results in nothing

There are three ways around it
1. Make Slic3r show the GUI by default, unless a switch like `--no-gui` is used, __or__
2. Edit https://github.com/alexrj/Slic3r/blob/66074f53989d312e8447c4de504d0602fd16ff42/package/linux/appimage.sh#L57 to include --gui, __or__
3. Add `Console=true` to the desktop file (resulting in a terminal window to open rather than the GUI)

probonopd, 2017-09-02:
Why would you do both? I'd say only add `--gui` and that should be it.

lordofhyphens, 2017-09-02:
Because the default is to fire up the gui if everything works and you don't have CLI arguments. If something is broken re: gui then it defaults to the CLI interface.

Can you pass --gui to invoking the AppImage (from a terminal) and see what gets spit out?

probonopd, 2017-09-02:
If I launch the AppImage with `--gui`, then it works.

lordofhyphens, 2017-09-02:
Interesting. Then adding it to appimage.sh should be fine then.

probonopd, 2017-09-02:
Indeed. :+1:

lordofhyphens, 2017-09-02:
I've added it; which required adding a new UI flag (in case you want to run the AppImage in CLI mode): `--no-gui`

lordofhyphens, 2017-09-02:
Closed via d6469b28.

closed 2017-09-02 by lordofhyphens

probonopd, 2017-09-02:
Thank you. Can you trigger a build on https://dl.slic3r.org/dev/linux/ please?

lordofhyphens, 2017-09-02:
It's triggered on pushing commits, so it's done already.

lordofhyphens, 2017-09-02:
https://travis-ci.org/alexrj/Slic3r/builds if you need to check the Linux build state in the future.

probonopd, 2017-09-02:
What would be awesome would be a link to the latest stable and the latest continuous build that would redirect to the respective file... that would help for the AppImageHub central directory of available AppImages, and other app collections.

lordofhyphens, 2017-09-02:
@alexrj would need to sort  that out, as the webserver is under his  control. You should open a new issue for that.

There won't be an AppImage of 1.2.9 ever though.

probonopd, 2017-09-03:
https://dl.slic3r.org/dev/linux/Slic3r-1.3.0-dev-c84bc71-x86_64.AppImage still does not launch the GUI... 

https://travis-ci.org/AppImage/appimage.github.io/builds/271433211#L596

probonopd, 2017-09-03:
Now it doesn't show the GUI anymore even when the AppImage is invoked with `--gui` on the command line. At least this worked before.

lordofhyphens, 2017-09-03:
Oh boy, wonder what I broke there. :/

lordofhyphens, 2017-09-03:
Should be fixed as of next build