slic3r/Slic3r #4112: Permalink to the latest stable and the latest continuous builds

probonopd, 2017-09-02:
What would be awesome would be a link to the latest stable and the latest continuous build that would redirect to the respective file... that would help for the AppImageHub central directory of available AppImages, and other app collections.

lordofhyphens, 2018-02-25:
I can get the continuous build in a relatively straightforward manner (adding the symlink during sftp).

lordofhyphens, 2018-02-25:
Should work now for Linux and OSX dev builds. I'm not sure it's particularly useful for the Windows folks

lordofhyphens, 2018-02-25:
Added to Linux builds as of 2d9580edb4c0e0ea157879ddf1d781460d1bbfa8

OSX is currently done through the Jenkins server, so no commit for that.

closed 2018-02-25 by lordofhyphens

lordofhyphens, 2018-02-25:
https://dl.slic3r.org/dev/linux/Slic3r-master-latest.AppImage is the symlink to the latest AppImage