slic3r/Slic3r #4113: Frist Layer of support material missing when brim overlaps support

4ndreas, 2017-09-02:
### Version
Slic3r-master.2017.08.14.1313.a54ca92.64bit

### Operating system type + version
Windows 10 64 Bit 

### Behavior
When the brim of the material starts overlapping the support material the first layer of the support material vanishes:
Here the 5 mm brim overlaps with the support material and vanishes completly
[image]

this is with no brim:
[image]

this is with 4mm brim:
[image]

and finally with slic3r 1.29 and 5 mm brim everything works fine:
[image]

and here the config and stl.
[slicer Bug.zip](https://github.com/alexrj/Slic3r/files/1272576/slicer.Bug.zip)

Also the first layer of the support vanishes to me randomly if the support material is inside the object linke in a hole:
[image]

lordofhyphens, 2017-09-03:
Does this look related to #4034 to you?

4ndreas, 2017-09-03:
I don't think so the brim seems fine.

l29ah, 2018-02-05:
Reproduced with dc9d7ed, i observed a support pillar being printed in thin air.

Billiam, 2018-07-16:
Also occurring in 12aa06e (June build):

With a 2mm brim that doesn't contact the supports:
[image]

With a 6mm brim that contacts some supports. Bottom layer disappears, and supports are unsupported.
[image]

Trying a range of settings between those two, the issue occurs when the separate brim around the supports contacts would contact the brim around the part.