slic3r/Slic3r #4114: Feature request: modifier to set Vase mode

eagleapex, 2017-09-03:
I love Slic3r. I am working on a model now that would print great with vase mode, but need structural support on the bottom to work. I would like to be able to have the bottom of the print as normal, but switch to vase mode for the remaining Z height of the model. A modifier STL would be a great way to do this.

lordofhyphens, 2017-09-04:
Vase mode is pretty intrusive when it comes to settings. You can, however, set more bottom solid layers to get that support.

EvilGremlin, 2019-10-19:
I also want this future badly. So bottom part will be printed as regular object, including top infill pattern, and then it switches to vase mode. 
Shouldn't be hard as vase mode is but a gcode postprocessor that works with external perimeter. But i'll need to redefine top layer location to be just below vase modifier and rewrite all setting checkups. Can you point me where i need to modify sources (so i won't miss something)?