slic3r/Slic3r #4116: MAGIC PRIME TOWER only for dual

studiodyne, 2017-09-04:
Feature requested : A simple idea that can change life of all

> Today , no software are ready for multicoloring , just Slic3r prusa i3 , but this version need to modify ' fw retract , and have some limitations of nozzles '

I suggest an idea , to have the possibility to have our own tower and in any color we want , but for this , the devs must make something for us

*    A  ' Prime tower ' smart functionnality
*    A choosen STL file is placed on the plate  , can be moved , scaled in any size to be what we need to   have the perfect volume for purging extruders
*   The STL file can be set with perimeters and fill 
*   The STL file is divided in 3 areas for 4 extruders , or 2 areas for 3 extruders , or 1 area for 2 extruders

*   The slicer  slice them but with some conditions 
   The STL is always print before layers !!!    
   if a layer finish , the current tool must update all no leveled zone of the STL in totality 
   if a tool change appears   the purge is made on  a free area 

With this system 

- NO WASTED MATERIALS
- All material is used to make an OBJECT !! Yeah !
- we can use the tools for scaling and moving ' no need of programmation' 
- we can stop the prime at the height we need if color dont change at a certain level
- We are full compatible with an illimited numbers of colors or extruders
- Multicolor priming is not easy to code , but with my system all is simplified and will be a standart for the future

This system is THE FUTURE OF MULTICOLOR PRINTING 
Hopefully SLIC3R will be the first to have this ecologic and smart functionnality

lordofhyphens, 2017-09-04:
Pull requests and algorithms get done faster.

lordofhyphens, 2017-09-04:
Looks like this duplicates #2778 #2894 and partially duplicates #3482.

studiodyne, 2017-09-05:
Yes my system prepare the future implementation of Multicolor tower
A dual tower is just a forced toolchange indication at layer change and toolchange conditionnals
But in an illimited number of colors , it's not the same schematic , then , the best is to create a general system , and if possible to recycle the wasted plastic

I think my idea is very easy to make , it's just some conditionnals , and a little mathematic , but if you cut the area by x only , or y only , it's not very mathematic

I hope this idea will interest the devs ,   very easy to make
The design of the tower is imported
The scaling is used by the toolbox of the software
The print toolpath is made by the slicer

It's just an algorithm that organize all
It's not like to create a visual interface with 3d rendering lol

studiodyne, 2017-09-05:
@lordofhyphens 
I have a question , that can be a ' too curious ' question , then you can delete this comment if needed

I have seen the PRUSA I3 multicolor version of slic3r , and i'm asking myself about this many features so perfect and not included in the last release of original slicer
Preview extruder color
Prime tower for 4 extruders
and more

Why  this features and this big job is not ported on SLIC3R original  ?
May it's done and wainting to be released , i don't know 
Or may be , not a very good collaboration between prusa team  ?

studiodyne, 2017-09-05:
Lol look , the prusa I3 multimat  have the same bug you have fixed 
When no FWretract , the tool change make the long retract
But if FW retract enable ,  tool change long retract is NOWHERE loll FORGOTTEN loll 
3 choices  
'implement the G10 S1 ' for all flavors
Lor leave the analogic long retract and recover G1 E-xxxxx  
Or specify FW retractation works ONLY with one extruder

But or MARLIN the problem is solved :smile:

G1 X90.859 Y106.414 F7200.000
G1 F1680
G1 X93.555 Y109.110 E14.65619
G1 E-41.00981 F3000.00000
G92 E0
G1 Z0.700 F7200.000
T1
G92 E0
G1 E-4.00000 F3000.00000
G92 E0

lordofhyphens, 2017-09-05:
The person paid to do work for Prusa3d doesn't have time to submit things
upstream (and be isn't being paid to do so). Anything that makes it up is
usually because it was ported up by someone else. Also their codebase has
diverged enough to make it a pain to pull his work in.

studiodyne, 2017-09-05:
👎  hard to hear that , but it's the reality 
They use your job , that have cost so many year and time , and works on it only for us , with an heterodyne code that need to be rewritten to work on all flavor
:cry:

lordofhyphens, 2017-09-05:
They are complying with the license. Everything else is a bonus.

lordofhyphens, 2018-01-19:
@studiodyne libslic3r (which is the place you would want to implement this) is C++. 

http://manual.slic3r.org/libslic3r-doc/ I have been working at improving this as I go.

closed 2018-02-18 by Studiodyne