slic3r/Slic3r #4118: Feature request

wjflier, 2017-09-04:
It would be great to be able to move the model in the z direction, especially for two extruder systems.
Unless I'm missing something.

lordofhyphens, 2017-09-04:
Can you describe why you need it more clearly?

wjflier, 2017-09-05:
example would be placing an object on the top of another item and print in two colors, such as a butterfly on a box, or one item i redesigned was an extruder block with and existing bowden fitting. 
[image]  here I placed the block on top of the fitting to make one stl.  This was done in Craftware.  I prefer Slic3r and think this could be a great addition.

lordofhyphens, 2017-09-05:
How about in the Object Settings panel, with the same mechanics as moving modifier meshes?

[image]

lordofhyphens, 2017-09-05:
https://ci.appveyor.com/project/lordofhyphens/slic3r/build/1.3.0-move-object-mesh-1367 

See the pull request when the build completes, this should show up in a build in https://dl.slic3r.org/dev/win/branches

wjflier, 2017-09-05:
That works, I thought it was there somewhere.  Appreciate the help.

lordofhyphens, 2017-09-05:
It was while I was first implementing moving modifier meshes.

Everyone at the time thought that there wasn't a use for being able to move
the parts too.

wjflier, 2017-09-05:
I combine stls a lot.  And I prefer Slic3r.  It's nice to have the ability in Slic3r to mod parts.
Thanks

PR #4119 (2017-09-05): Move part meshes around in Object Settings.
Re-added the ability to move part meshes relative to another part mesh in the same object.

Fixes #4118
closed 2017-09-05 by lordofhyphens