slic3r/Slic3r #4120: GUI colour scheme is hard/impossible to see

BETLOG, 2017-09-05:
### Version
slic3r-1.3.0-dev-8aff101-linux-x64

### Operating system type + version
Description:    Ubuntu 14.04.5 LTS
Codename:       trusty
Linux betlogbrick 3.13.0-129-generic #178-Ubuntu SMP Fri Aug 11 12:48:20 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
* GUI colour scheme is hard/impossible to see
* Window decorations: Oxygen, Cursor Theme: Oxygen White, Desktop Theme: Oxygen, Color Scheme: default.
* Expected: GUI complies with workspace settings, or uses enforced theme that is easily visible.
* Actual: GUI either uses some unusual combination of workspace settings and enforced settings, or it's enforced theme is quite badly chosen, resulting in many items being slightly difficult or impossible to discern, and looking like something is wrong with the icon/menu.
  * Screenshots: 
[image]

I didn't see any other issues relating to this, but I can't imagine nobody has mentioned it before.

lordofhyphens, 2017-09-05:
Wow. 

AFAIK we shouldn't be specifying color, but I guess I get to go looking again (there had been an issue with the text on the host log and dark themes).

lordofhyphens, 2017-09-05:
Is this the appimage or the archive or did you build from source?

BETLOG, 2017-09-05:
Regular browser download from the link: https://dl.slic3r.org/dev/linux/slic3r-1.3.0-dev-8aff101-linux-x64.tar.bz2

archer46, 2022-09-06:
This is still an issue. Has any progress been made?