slic3r/Slic3r #4121: Problem in CPP --export-svg option

sauvik3, 2017-09-07:
Version

1.3.0 dev

Operating system

Windows 7 Professional SP1 x64

Build Configuration

Boost 1.63.0 x64
Visual Studio 2015 Professional x64

Runtime Error

Running --export-svg option causes slic3r.exe to get hanged.
From surface, it looks like 'layers' keeps on growing and does not stop.
This is SLAPrint.cpp, Line 41.
There is same behavior on MinGW compilation also.

sauvik3, 2017-09-08:
[Slic3r.zip](https://github.com/alexrj/Slic3r/files/1286929/Slic3r.zip)

lordofhyphens, 2017-09-08:
Need a copy of a 3D Model you were seeing this with.

sauvikroy, 2017-09-11:
[test.zip](https://github.com/alexrj/Slic3r/files/1291492/test.zip)
Here is the model I used. Sorry for late, didn't have access to computer.

sauvikroy, 2017-09-11:
The perl version however seems to work correctly to generate SVG file.

sauvik3, 2017-09-11:
Here is the file. Sorry for late, didn't have access to computer. The perl version however seems to work correctly for the SVG export version.
[test.zip](https://github.com/alexrj/Slic3r/files/1291501/test.zip)

lordofhyphens, 2017-09-29:
@sauvik3 Are you supplying a configuration (at least a layer height) to the CLI? Currently I've noticed that it will hang forever if the layer height is 0. I'm still looking into it in #4139 and will update that issue as I figure out what is going on. 

Also note #4130.

closed 2017-09-29 by lordofhyphens

sauvik3, 2017-09-30:
I did not supply any explicit parameters. I used the default configuration only.

sauvik3, 2017-09-30:
The issues you raised are the same as mine, better documented. :)