slic3r/Slic3r #4123: program crash when bed dimensions are zero

arnottcr, 2017-09-09:
### Version
1.2.9

### Operating system type + version
Linux: Arch Linux

### Behavior
* when setting the bed shape, if you ever change the value to `0`, the program crashes with the following error:
```
Illegal division by zero at /usr/share/perl5/vendor_perl/Slic3r/GUI/BedShapeDialog.pm line 274.
```
* this happens during the quickstart wizard, and if you ever edit the bed settings
  * you should be able to fix it by just ignoring zero values in these cases

lordofhyphens, 2017-09-09:
Please try to reproduce on 1.3.0-dev. (https://dl.slic3r.org/dev)

lordofhyphens, 2017-09-09:
This was fixed some time ago. #3681

closed 2017-09-09 by lordofhyphens

arnottcr, 2017-09-09:
yeah, I tried to search through old issues, but may have missed something. I cannot reproduce on 1.3.0, so closure is the right call; thanks for the help.

lordofhyphens, 2017-09-09:
I had searched for "bed crash" in closed issues, if that helps.

arnottcr, 2017-09-09:
I got hung up on the "Illegal division by zero", as that is printed to stdout