slic3r/Slic3r #4124: Bridge not recognized as bridge.

rewolff, 2017-09-10:
### Version
1.3.0-dev
1.2.9-1035-gfbb606d

### Operating system type + version
ubuntu-16.04.3

### Behavior
 (I'll try to attach the STL, I'll add a link if adding the STL doesn't work) 
The bridges (strings) in the model are not recognized as bridges. They are extruded at the same speed as the "outer perimeters".  I have created a profile with a unique speed for bridges. 

The strings are 0.2mm high, so slice with 0.2mm layer height. 

I expected to be able to tune the bridges with the settings for bridges but that is not possible. 

For these bridges I edited the resulting G-code and created what I call "anchors": The extrusion command starts and stops 1mm further inside the object. Maybe if detecting the bridge correctly then this would have happened automatically. But the print quality is MUCH better after the edit. 

[harp_test.zip](https://github.com/alexrj/Slic3r/files/1290339/harp_test.zip)

I have included the edited gcode file. I used octoprint to view the gcode, but you guys are probably more familiar with slic3r itself to do that. :-)

I recognize the bridges in the G-code by looking for Y0.000 

Maybe as a generic request: often the print quality suffers a bit on a start/stop of the extrusions. So in critical places, you don't want that to happen. I've had a "taper" on an object where the final "single extrusion width" part would be extruded separately on each layer. This resulted in a separate tower from the main object, just because my printer started extruding a bit late. In that case, placing one of the endpoints of the "trace the perimeter" on the taper would have worked. 

In the example here, a single tour around the perimeter is possible, starting and ending near the "bridge" to the other part.

lordofhyphens, 2017-10-01:
As a FYI: Octoprint won't give you much information about the actual print line widths.

lordofhyphens, 2017-10-02:
[image]

So here's a picture of what the top string's layer looks like after process() completes. As you can see, PerimeterGenerator 

As far as I can tell the strings are being picked up by the thin wall detection.

Bridges also don't start to be "detected" until there's room to put infill in there. 

What might be feasible is to do similar as to what you did and do thin walls first, overlapping the previous layer (anchors) and then draw the perimeters over them.

rewolff, 2017-10-02:
That picture, is that of "a slice of the object" or of a generated (preliminary) tool-path? The tool path should have a beginning and an end. I can't see it in the picture.  The tool path should not follow the perimeter exactly but it should be offset by about 0.2mm (for my 0.4mm nozzle). As the string is 0.4mm wide, that would place the two perimeter toolpaths exactly on top of each other. I don't see that.