slic3r/Slic3r #4127: Unable to change perimeter gap

albabis, 2017-09-17:
### Version
1.2.9

### Operating system type + version
Opensuse Linux Leap 42.2

Hi
Whatever settings i am using the distance between perimeter lines is always around 0.535 mm.
My head is 0.3mm so the perimeter lines are not even touch each other. 

[STL_Config.zip](https://github.com/alexrj/Slic3r/files/1308886/STL_Config.zip)

lordofhyphens, 2017-09-17:
Please check with 1.3.0-dev https://dl.slic3r.org/dev/linux and add a
screenshot of the 3d preview, showing what you are talking about.

The spacing between adjacent perimeters is sized for the width of the
traces.

Nominally a 0% overlap between perimeters and infill (advanced print
settings) will try to put the two next to each other, but often the lines
don't work out that way.

albabis, 2017-09-17:
A correction to my first message, my head is 0.4. Sorry for that.

The preview looks always the same.

Look at the gcode in file test.gcode and test1.gcode. The only difference is that i change the head diameter in configuration. I set it 0.8 in test.gcode and 0.3 in test1.gcode.

In both files the first perimeter line start from X103.629 Y103.629 (line 50 in files) and end in X96.371 Y103.629 (line 53). The second parallel to the first perimeter line starts at X104.164 Y104.164 (line 57) and end at X95.836 Y104.164 (line 59).

The two lines spaced 0.535mm from each other but the extruded string is around 0.42-0.45 (its hard to have exact measurement)

I did most of the tests by just produce and check the gcode. In real life i did only three tests by changed the head diameter in configuration (0.4 0.3 0.5). 

I try to play with several parameters in the configuration file trying to produce a different spacing, finally i succeed by changed the [First Layer] option. I have not trying it yet to the printer.
[files.zip](https://github.com/alexrj/Slic3r/files/1309271/files.zip)

lordofhyphens, 2017-09-17:
* Set your nozzle diameter configuration option to what it really is on your printer. You're on your own with any other configuration setting.
* Your extrusion width is set to be below the nozzle diameter. This is generally considered to be unreliable and/or experimental. Pick an extrusion width larger than your nozzle. That the first layer prints at all is because Slic3r defaults first layer extrusion width to 200% of the layer height, which gets you to around 0.6mm (first layer height in this case). 
* It'd help with figuring out what is going on if you set your printer to use volumetric E (Printer settings); then you can work out the exact volume that Slic3r says it wants to extrude. You'll need to configure your printer to support volumetric E if you want to print with that gcode.
    * It also helps to turn on verbose gcode (Print settings).