slic3r/Slic3r #4128: crash on sending to duet - with fix, please apply

rob-miller, 2017-09-17:
### Version
_slic3r-1.3.0-dev-1a2df5e_ at least

### Operating system type + version
_OX X El Capitan 10.11.6_

### Behavior
* _Slic3r crashes on attempting to send to printer when configured for Duet printer_
* _steps to reproduce: Configure for duet, load .stl file and 'send to printer'; should crash even if no duet for testing_
* _Expected: successful send_
* _Actual: crash_

In Plater.pm:send_gcode() the line

open my ($gch,$path);

is in error.  Please change to

open my $gch, $path;
or 
open (my $gch,$path);
if your coding standard requires the parens.

Sorry I am unable to submit as a pull request, I have not yet configured to develop Slic3r as I would like to.

Thank you,
rob.

lordofhyphens, 2017-09-24:
I'm mostly curious as to why it is crashing for you; which perl version?

lordofhyphens, 2017-09-24:
https://perldoc.perl.org/functions/open.html indicates that `open (my $gch,$path);` is correct, so we'll go with that.

closed 2017-09-24 by lordofhyphens

rob-miller, 2017-09-28:
Sorry, did not see the query above - I believe the problem is that the syntax

my ($gch, $path);

creates a new and undefined local variable $path.   This overrides the previous definition of $path, so not reading the intended gcode data and not sure where it fails after that.  

It seems like this should have crashed in any test environment?  In my testing the fixed behaviour without the Duet available is to pop up a timeout window.

Thank you for fixing.

commit 5ae6a7fe (2017-09-24): Corrected open() syntax. Fixes #4128, thanks to @rob-miller for the heads-up.