slic3r/Slic3r #4130: Floating point exception thrown by Slic3r in DLP mode.

lordofhyphens, 2017-09-21:
Commit: 7e018f1d381d6b64297abb092399b2a29adf3526
Error: Program terminated with signal SIGFPE, Arithmetic exception.

Core-file: 
````
    out=out@entry=0x7f5bd6ffcb80) at src/libslic3r/Fill/FillRectilinear.cpp:528
#1  0x00007f5bf1e1a121 in Slic3r::Fill::fill_surface (this=this@entry=0x7f5bc4001530, surface=...) at src/libslic3r/Fill/Fill.cpp:70
#2  0x00007f5bf1d3151f in Slic3r::SLAPrint::_infill_layer (this=<optimized out>, i=<optimized out>, _fill=<optimized out>) at src/libslic3r/SLAPrint.cpp:195
#3  0x00007f5bf1d352e7 in boost::function1<void, unsigned long>::operator() (a0=1, this=0x7f5bd6ffcee0) at /usr/include/boost/function/function_template.hpp:771
#4  Slic3r::_parallelize_do<unsigned long>(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>) (queue=0x7fffdc70b5a0, queue_mutex=0x7fffdc70ae90, func=...) at src/libslic3r/libslic3r.h:90
#5  0x00007f5bf1d33ec9 in boost::_bi::list3<boost::_bi::value<std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*>, boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void (unsigned long)> > >::operator()<void (*)(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>), boost::_bi::list0>(boost::_bi::type<void>, void (*&)(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>), boost::_bi::list0&, int) (
    a=<synthetic pointer>..., 
    f=@0x5624b0e49c58: 0x7f5bf1d35260 <Slic3r::_parallelize_do<unsigned long>(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>)>, this=0x5624b0e49c60) at /usr/include/boost/bind/bind.hpp:398
#6  boost::_bi::bind_t<void, void (*)(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>), boost::_bi::list3<boost::_bi::value<std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*>, boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void (unsigned long)> > > >::operator()() (this=0x5624b0e49c58) at /usr/include/boost/bind/bind.hpp:1294
#7  boost::detail::thread_data<boost::_bi::bind_t<void, void (*)(std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*, boost::mutex*, boost::function<void (unsigned long)>), boost::_bi::list3<boost::_bi::value<std::queue<unsigned long, std::deque<unsigned long, std::allocator<unsigned long> > >*>, boost::_bi::value<boost::mutex*>, boost::_bi::value<boost::function<void (unsigned long)> > > > >::run() (this=0x5624b0e49aa0)
    at /usr/include/boost/thread/detail/thread.hpp:116
#8  0x00007f5bf1320116 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.62.0
#9  0x00007f5bf3bca494 in start_thread (arg=0x7f5bd6ffd700) at pthread_create.c:333
#10 0x00007f5bf390caff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
````

jdevelop, 2017-09-21:
```
      Message: Process 12092 (perl-local) of user 1000 dumped core.
                
                Stack trace of thread 12152:
                #0  0x00007f054b01c343 _ZN6Slic3r5Point13align_to_gridERKS0_S2_ (XS.so)
                #1  0x00007f054b029a53 _ZN6Slic3r13FillHoneycomb20_fill_surface_singleEjRKSt4pairIfNS_5PointEERNS_9ExPolygonEPSt6vectorINS_8PolylineESaIS9_EE (XS.so)
                #2  0x00007f054b0a52f6 _ZN6Slic3r4Fill12fill_surfaceERKNS_7SurfaceE (XS.so)
                #3  0x00007f054b0a653f _ZN6Slic3r8SLAPrint13_infill_layerEmPKNS_4FillE (XS.so)
                #4  0x00007f054b0a8203 _ZN6Slic3r15_parallelize_doImEEvPSt5queueIT_St5dequeIS2_SaIS2_EEEPN5boost5mutexENS8_8functionIFvS2_EEE (XS.so)
                #5  0x00007f054b0a7f5d _ZN5boost6detail11thread_dataINS_3_bi6bind_tIvPFvPSt5queueImSt5dequeImSaImEEEPNS_5mutexENS_8functionIFvmEEEENS2_5list3INS2_5valueIS9_EENSI_I
                #6  0x00007f054b109295 thread_proxy (XS.so)
                #7  0x00007f054d45008a start_thread (libpthread.so.0)
                #8  0x00007f054ca001bf __clone (libc.so.6)
                
                Stack trace of thread 12094:
                #0  0x00007f054c9f5cbb __poll (libc.so.6)
                #1  0x00007f0546ee5c09 n/a (libglib-2.0.so.0)
                #2  0x00007f0546ee5d1c g_main_context_iteration (libglib-2.0.so.0)
                #3  0x00007f0546ee5d61 n/a (libglib-2.0.so.0)
                #4  0x00007f0546f0cae5 n/a (libglib-2.0.so.0)
                #5  0x00007f054d45008a start_thread (libpthread.so.0)
                #6  0x00007f054ca001bf __clone (libc.so.6)
                
                Stack trace of thread 12153:
                #0  0x00007f054ca001b1 __clone (libc.so.6)
                #1  0x00007ffc4693a750 n/a (n/a)
                #2  0x00007ffc4693a601 n/a (n/a)
                #3  0x0000000000000000 n/a (n/a)

```

lordofhyphens, 2017-09-21:
Ran the minimal CLI with --export-svg on a debug build, got a coredump.
````
./slic3r ~/Downloads/E3D_V6_fan.STL --export-svg
````
````
0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007f3dbd2fc3fa in __GI_abort () at abort.c:89
#2  0x0000560b10b7e18d in __gnu_cxx::__verbose_terminate_handler() ()
#3  0x0000560b10af6c46 in __cxxabiv1::__terminate(void (*)()) ()
#4  0x0000560b10af6c91 in std::terminate() ()
#5  0x0000560b10af6bd9 in __cxa_throw ()
#6  0x0000560b10af691c in operator new(unsigned long) ()
#7  0x0000560b109eec9e in __gnu_cxx::new_allocator<Slic3r::SLAPrint::Layer>::allocate (this=0x7ffc4ee112a8, __n=33554432)
    at /usr/include/c++/6/ext/new_allocator.h:104
#8  0x0000560b109edb0e in std::allocator_traits<std::allocator<Slic3r::SLAPrint::Layer> >::allocate (__a=..., __n=33554432)
    at /usr/include/c++/6/bits/alloc_traits.h:436
#9  0x0000560b109ec776 in std::_Vector_base<Slic3r::SLAPrint::Layer, std::allocator<Slic3r::SLAPrint::Layer> >::_M_allocate (this=0x7ffc4ee112a8, __n=33554432)
    at /usr/include/c++/6/bits/stl_vector.h:170
#10 0x0000560b109ea500 in std::vector<Slic3r::SLAPrint::Layer, std::allocator<Slic3r::SLAPrint::Layer> >::_M_emplace_back_aux<Slic3r::SLAPrint::Layer>(Slic3r::SLAPrint::Layer&&) (this=0x7ffc4ee112a8, __args#0=<unknown type in /mnt/fiver/slic3r/src/slic3r, CU 0x31f336, DIE 0x371b2c>) at /usr/include/c++/6/bits/vector.tcc:412
#11 0x0000560b109e85be in std::vector<Slic3r::SLAPrint::Layer, std::allocator<Slic3r::SLAPrint::Layer> >::emplace_back<Slic3r::SLAPrint::Layer>(Slic3r::SLAPrint::Layer&&) (this=0x7ffc4ee112a8, __args#0=<unknown type in /mnt/fiver/slic3r/src/slic3r, CU 0x31f336, DIE 0x371b2c>) at /usr/include/c++/6/bits/vector.tcc:101
#12 0x0000560b109e6a1c in std::vector<Slic3r::SLAPrint::Layer, std::allocator<Slic3r::SLAPrint::Layer> >::push_back(Slic3r::SLAPrint::Layer&&) (
    this=0x7ffc4ee112a8, __x=<unknown type in /mnt/fiver/slic3r/src/slic3r, CU 0x31f336, DIE 0x370cc6>) at /usr/include/c++/6/bits/stl_vector.h:933
#13 0x0000560b109e13f5 in Slic3r::SLAPrint::slice (this=0x7ffc4ee11190) at /home/lenox/slic3r/xs/src/libslic3r/SLAPrint.cpp:42
#14 0x0000560b1091d17f in main (argc=3, argv=0x7ffc4ee11ac8) at /home/lenox/slic3r/src/slic3r.cpp:134
````

lordofhyphens, 2017-09-29:
Well, part of the problem is that config.layer_height defaults to 0 apparently, so the layer push in the slice() method adds layers until it runs out of memory.

lordofhyphens, 2017-09-29:
@jdevelop on the CLI, anyway, you probably want to specify the options manually. 

for the GUI I suspect the config isn't being loaded.

lordofhyphens, 2017-09-29:
@jdevelop as a short-term workaround, try setting your infill to 100% and rectilinear. Or 0% infill.