slic3r/Slic3r #4132: huge gaps at seam with volcano nozzle

cameronbergh, 2017-09-22:
### Version
_1.3.0-dev_ but this also happens with 1.2.9

### Operating system type + version
windows 10, but also tested on macOS sierra

### Behavior
* _Describe the problem_
since upgrading to the 1.2mm volcano nozzle ive been noticing some gaps in the perimeters of my prints, as if the nozzle starts moving but the extruder isnt extruding at the seam.

things tested:
i thought maybe desabling retraction would help, but it makes no difference. 
i set pressure advance to .05, makes no difference. 
setting pressure advance to 1, makes gaps and huge blobs elsewere.

after trying a little investigating, i found that the gaps appear in the repetier gcode visualizer.

here is how it appears in slic3r:
[image]

here is how it appears in repetier with seams aligned:
[image]

this is how repetier renders it when i enable random seams:
[image]

how it actually prints (random seams):
[image]

how it prints (2 examples) (seams aligned):
[image]
[image]

#### STL/Config (.ZIP) where problem occurs

[Slic3r_config_bundle.zip](https://github.com/alexrj/Slic3r/files/1323457/Slic3r_config_bundle.zip)

lordofhyphens, 2017-09-22:
Could you upload a config export, not a bundle? Bundles are for backing up
settings, single exports are for reproducing settings.

I really need to get AMF project files working on Slic3r to make it easier
to handle this.

If it is showing up in your exported gcode (Do not trust RepHost preview,
it can't account for width), I would confirm the extrusion width. 1.2mm is
pretty big for a lot of things.

cameronbergh, 2017-09-22:
here is my config file!

[config.zip](https://github.com/alexrj/Slic3r/files/1323844/config.zip)

The nozzle is indeed the 1.2mm E3D volcano nozzle.

I am now using Cura, and the gapping doesnt seem to be happening, so, i feel confident that this isnt a hardware problem.

Thank you for your help.
Keep up the good work!

-cameron

lordofhyphens, 2017-09-24:
Is there ooze or something else going on? A loss of pressure would generate those gaps, and I know that volcanoes can be hard to keep pressure on.

cameronbergh, 2018-03-20:
the pressure issue is more of a hardware issue, this is/was a bug with slic3r, also perhaps i wrote the wrong title for this bug report since at the time i didn't really understand what was happening. 

 if you look closely at the screenshots you can see near the bottom of the part where the outer layers are not being printed. in the slic3r view thing it would render the lines, but they would not print. the missing lines are rendered in the repetier simulator as shown in the screenshots. i've since been using cura and the problem is gone.

**_the blue printed part photo is the unrelated pressure issue._**

 in the photo of the actual printed part (white and/or clear), the missing lines appear to be sorta there because i was messing with the pressure settings 

**the real thing to take note of is the difference between the way repetier renders the part and how slic3r renders the part in the screeenshots**

thanks for making an awesome program!