slic3r/Slic3r #4134: Export does not lock the output file. (gcode)

Careknight, 2017-09-24:
### Version
1.2.9.41

_Use `About->About Slic3r` for release versions._
Generic Export
_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7 64bit

### Behavior
When exporting to file, Slic3r is not locking the file correctly. An OS copy action can be performed while the file is still being written too. It would appear that performing the copy and paste while slic3r is writing blocks slic3r from completing the write of the file. This results in a malformed file that will print, but will not complete the whole object.

###reproduce
I believe this is a generic issue and therefore I shall not provide a config or files.
If it turns out not to be generic then I shall.
A) Observe the writing to file and the total size of the object,.
1. Import a large STL into slic3r (My object was about 120x40x40mm.
2. Open File explorer to the location where the file will be exported. (Make sure that "File Size" is visible.)
3. Using Slic3r, export you print to the location from 2.
4. Focus the explorer browser and repeatedly click refresh, The size of the file will slowly increase.
B) Malform the output file
1. repeat Steps 1-3 but export to a different name to keep your reference file.
2. Before the total file size has been achieved, perform a copy Paste within the folder.

You will now have two files, one of which is correctly formed, the other is significantly smaller. 
Printing with the small file will result in a print that will stop part way through the print.
I get the impression that homing at the end of print is a feature of the Gcode , but this will give the same effect as described by a "Faulty SD Card" symptom. 
It is in the domain of the user to make sure that the copy from HDD to SD completes correctly, however Slic3r put the output file into a state where it cannot be processed until the write to file is complete.

In fact it may even be that this issue is only being hit after a duration of use as the size of build probably increases with confidence of using the 3D printer.

###Suggestion
Write to temp file and then rename at the end of the write operation.

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

lordofhyphens, 2017-09-29:
Can you give the build in the pull request a try when it completes? I set file locking which (hopefully) will work. Temp files are less than reliable as you have the same problem with people mucking with it.

Careknight, 2017-09-29:
No problem. Will be this evening

Careknight, 2017-09-30:
I have test the behavior of the temp file which eliminates the garbled file behavior. I shall keep an eye on the pull request when it's in to test the lock behavior.

lordofhyphens, 2017-09-30:
Well nevermind then if the current master is creating a temp file on windows. 
I had thought you were using a current version of -dev before. 

I'll just close the lock/unlock PR (which failed to package/build on Windows anyway).

closed 2017-09-30 by lordofhyphens