slic3r/Slic3r #4137: This machine config breaks Slic3r

Kolodner, 2017-09-25:
### Version
_I've tried both 1.2.9 and the 1.3.0-dev builds and the Prusa version so far_

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_Windows 10 v1703_
_Sager NP9170 laptop with a GTX 680M with current drivers_

### Behavior
* _On attempting to load the following machine definition, Slic3r crashes in all the versions I've tried. This used to work at a previous point in time, so I'm not sure what has broken along the way:_
[https://www.dropbox.com/s/b38bbwlumxtovs5/Slic3r_config_DVT4_20150812.ini?dl=0](url)
* _Steps to reproduce:_
1) Start a clean version of Slic3r
2) Try to load this config file. 
* _Expected Results_
It would load the machine parameters.
* _Actual Results_
It Crashes
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/
No.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
I've included a link to the file in my Dropbox above, didn't bother zipping because it's so small.

lordofhyphens, 2017-09-25:
Can you just zip the config and drop it into the web interface for github?

lordofhyphens, 2017-09-25:
That's a STL bed shape, isn't it?

I can't reproduce with https://dl.slic3r.org/dev/win/Slic3r-master.2017.09.24.1383.5514c84.64bit.zip

Kolodner, 2017-09-25:
I believe that's in there, yes. It's got a rectangular build area over a circular base plate, so the front two corners overhang the plate, and have to be trimmed in (which the manufacturer did for the config)

Since you can't reproduce on your end, what can I send you as far as crash-dumps and stacks that would be helpful (and how would I do it, not a developer here).

Kolodner, 2017-09-25:
[Slic3r_config_DVT4_20150812.zip](https://github.com/alexrj/Slic3r/files/1330892/Slic3r_config_DVT4_20150812.zip)

lordofhyphens, 2017-09-25:
Did you confirm with the linked version of the build? That's right off of the release.

Kolodner, 2017-09-26:
Yep, still crashes on trying to load that config:

[image]

lordofhyphens, 2017-09-26:
Can you delete the Bed shape line and see if it still crashes?

lordofhyphens, 2017-09-26:
I can't get it to crash on my Linux system either.

If you open up a command line window and run slic3r-console.exe it should spit out an error message when it crashes.

lordofhyphens, 2017-09-26:
Just also checked on my Windows 10 system with an nVidia 740M graphics card, no crash either :(

lordofhyphens, 2017-09-26:
That screenshot screams "OpenGL issue".

Kolodner, 2017-09-26:
hmmm. I'll try running the console & deleting the bed shape. Also, I double-checked my drivers and oh look, Microsoft had decided they wanted to install a different display adapter driver for my GTX680:
[image]
So, I'm trying to update to the latest Geforce driver first.

Kolodner, 2017-09-26:
That appears to have been it. When my video card drivers were updated, this _*appears*_ to be resolved, in that it  didn't crash on config import.

Thanks for pointing me to OpenGL as likely culprit, which prompted a drivers update. Not sure why it was rolled back to May's version as I'm pretty sure I'd updated more recently than that. (also not sure why it was working previously then suddenly not, if the driver had been the same this whole time and I've definitely been slicing things in June/July).

closed 2017-09-26 by lordofhyphens