slic3r/Slic3r #4141: Slic3r Won't connect to Zonestar 3D Printer

Witenite, 2017-09-30:
### Version
_Version 1.3.0-dev

### Operating system type + version
Ubuntu Linux 16.04 (fully up to date)

### Behavior
Similar to Slic3r won't connect to RUMBA board running marlin #3158
I can see /dev/ttyUSB0 (com device for the Zonestar 3D printer) however when I click on test (for any baud rate) I get a "Connection failed" error
I have tried various things to get the comms port to work, but slic3r refuses to operate correctly. Please help!

#### STL/Config (.ZIP) where problem occurs
Probably not relevant, as I am able to load an STL file without issue, just can't print it as I slic3r refuses to talk to the printer.

lordofhyphens, 2017-09-30:
Permissions?

lordofhyphens, 2017-09-30:
`ls -l /dev/ttyUSB0`, make a note of the permissions set and the group. Make sure you are in the group that owns `/dev/ttyUSB0` (usually dialout).

Witenite, 2017-09-30:
When I run ls -l /dev/ttyUSB0 I get the following:
crw-rw---- 1 root dialout 188, 0 Oct  1 09:43 /dev/ttyUSB0

As this says root, does this mean I don't have the required permissions? How do I set or change the permissions. Sorry for all the questions, but a thorough search on Ubuntu forums shows nothing in this regard (unusual as they are normally very good).

When I check groups, I appear to be in there:
graham adm cdrom sudo dip plugdev lpadmin sambashare

graham being my current login, so I presume I should have the required permissions in this regard.
Thanks for your assistance lordofhyphens.

lordofhyphens, 2017-10-01:
Ensure you are in the dialout group. A quick test for Slic3r software
issues would be to chmod the port to 666 (or run Slic3r as root, but I
would never recommend that in general).