slic3r/Slic3r #4145: Turn off auto-arrange / auto-centre in command line mode (CLI)

stepkurniawan, 2017-10-04:
Huge thanks for the support on my last question, especially to @lordofhyphens . I stumbled upon another difficulty here probably considering my lack of knowledge in the code design structure. 

### Version 1.3.0-dev
revision : 5134 (I'm using TortoiseSVN)

### Operating system: Windows 10
A bit similar to issue #404

### Behavior
So I'm using a different GUI to pipe in information to the slic3r. (I'm guessing I'm using the slic3r-console.exe instead of the slic3r.exe)
Everytime I slice my STL, the resulting G-code is always shows that the STL is in the centre of the print bed. I even tried slicing it when the position is far off the bed, but the g-code still the same. 
My guess is that there is an auto-centering / auto-arranging stl defaultly turned on in the slic3r command line mode. I tried turning it off in the GUI, but as you can guess already, it does not affect the command line mode because they are independent. 

My question is, how can I turn off (or on) the auto centering in the command line mode?
And out of curiosity, when I slice with command line, am I really using the slic3r-console.exe? 

Thank you for your attention!

Best regards, 
Stephen

lordofhyphens, 2017-10-04:
Your svn revision number makes no sense, as we use git.

Also:  --dont-arrange

lordofhyphens, 2017-10-04:
Let me know if that doesn't work for you. Otherwise I can close the issue :)

stepkurniawan, 2017-10-04:
Thanks once again. 
I tried the --dont-arrange. Now, it does not arrange the Y axis, but the program still automatically arrange the X and Z axis. The height now is automatically always making the STL touching the printbed. And the X axis is always shifted by minus <some constant> milimeter. 
btw, what is MD :D ?

lordofhyphens, 2017-10-04:
> is always shifted by minus milimeter.

Logic does currently force the print to be on the printbed. I can look into
seeing what changing this breaks.

lordofhyphens, 2017-10-04:
Initial workaround if you want to play around with a modified version is to remove the Z components of ModelObject::center_around_origin() and ModelObject::align_to_ground() in Model.cpp for the translations done.

lordofhyphens, 2017-10-04:
If you didn't want to recompile and just wanted it to affect the CLI, comment out line 94 in lib/Slic3r/Print/Simple.pm

stepkurniawan, 2017-10-06:
@lordofhyphens of course, that makes sense. The Z-axis arrangement is necessary, in every 3D Printer 
😄. My question remains on the X axis movement tho, but I will test your branch first, thanks!

lordofhyphens, 2017-10-23:
@stepkurniawan well?

stepkurniawan, 2017-10-24:
@lordofhyphens Sorry my workload suddenly forced me to focus on something else first. It will take a few weeks until I can try it. Sorry !! :((

expertmm, 2017-12-05:
that unexpected behavior seems to stem from this issue: #4190