slic3r/Slic3r #4147: Wrong G-code speed for MonoPrice Select Mini V2

jdevelop, 2017-10-06:
### Version
Built from sources, 1.3.0-dev
Commit: a28f431d044eca7ea709a3771c2858e35e733eec

### Operating system type + version

Arch Linux, Kernel 4.13.3-1-ARCH

### Behavior

I have sliced the simple model using both Slic3r and Cura, and I realized that for certain layers Slic3r does calculate the speed incorrectly. I set the max speed to 30mm/s, however in the generated GCode file there are bunch of statements like:

```
G1 F3360
G1 X68.128 Y60.288 E90.81899
G1 E90.46899 F3000.00000
G92 E0
G1 Z0.863 F4200.000
; Layer end
G1 X66.800 Y66.800 F4200.000
G1 E7.00000 F3000.00000

G1 F3000
G1 X47.200 Y66.800 E7.96790
G1 X47.200 Y47.200 E8.93580
G1 X66.800 Y47.200 E9.90370
G1 X66.800 Y66.740 E10.86864

G1 X66.486 Y66.486 F4200.000

G1 F3000
G1 X47.514 Y66.486 E11.80552
G1 X47.514 Y47.514 E12.74239
G1 X66.486 Y47.514 E13.67926
G1 X66.486 Y66.426 E14.61317
```

I have no overhangs or bridges, so habing the speed of **50mm/s** is weird.

I tried exactly the same STL file in Cura, and it generated the G-Code that doesn't have this problem:

```
G1 F1200 X51.715 Y69.138 E33.54118
G1 X51.785 Y69.208
G0 F2100 X52.337 Y69.138
G1 F1200 X50.859 Y67.66 E33.63082
G1 X50.789 Y67.59
G0 F2100 X50.859 Y67.037
G1 F1200 X52.96 Y69.138 E33.75825
G1 X53.03 Y69.208
G0 F2100 X53.582 Y69.138
G1 F1200 X50.859 Y66.415 E33.9234
G1 X50.789 Y66.345
G0 F2100 X50.859 Y65.793
G1 F1200 X54.204 Y69.138 E34.12628
G1 X54.274 Y69.208
G0 F2100 X54.826 Y69.138
G1 F1200 X50.859 Y65.171 E34.36688
G1 X50.789 Y65.101
G0 F2100 X50.859 Y64.548
G1 F1200 X55.449 Y69.138 E34.64527
G1 X55.519 Y69.208
G0 F2100 X56.071 Y69.138
G1 F1200 X50.859 Y63.926 E34.96138
```
Attached is ZIP with the STL, G-Code files and shots of the prints (just 3 layers).

lordofhyphens, 2017-10-06:
@jdevelop in the future include the exported config.ini (Ctrl+E) for Slic3r.

lordofhyphens, 2017-10-06:
Max print speed doesn't work like that; it's there to govern the maximum for *autospeed*. You need to set the print moves to auto to get a cap of 30mm/s. 

If you want to limit how fast the printer extrudes even if you have fixed print speed things (and you have bridges set to 50mm/s), you need to set a maximum volumetric speed. Or do it in your firmware (that's much more reliable for a max speed; IIRC the Monoprice fw supports it).

closed 2017-10-06 by lordofhyphens

jdevelop, 2017-10-06:
I am not using auto-speed at all, and the G-Code file contains all the settings I have.
Attaching the `config.ini` just in case.
[config.ini.txt](https://github.com/alexrj/Slic3r/files/1361338/config.ini.txt)

jdevelop, 2017-10-06:
My settings for the speed

[image]

lordofhyphens, 2017-10-06:
As I explained over IRC, the "max print speed" option isn't doing anything (hence why the option is disabled in your screenshot). If you want to limit print speeds, do it in mm^3/s and put it in "max volumetric speed). 

The "fast" prints you are seeing are because they are being classified as bridges (on account to being bridging over support material interface aka a raft).