slic3r/Slic3r #4149: Do not change layer at first tool change

carlonb, 2017-10-09:
### Version
version 1.2.9
### Operating system type + version
Windows 10
### Behavior
Do not apply a layer change (Z increment) at first tool change at line 565 as seen on gcode file annexed, other layer and tool change seems good.
Note: Before/After layer change and Tool change are empty only for gcode debugging, the same happen with full fields.
#### STL/Config (.ZIP) where problem occurs
[Colonna10x10-4colori.zip](https://github.com/alexrj/Slic3r/files/1368120/Colonna10x10-4colori.zip)

platsch, 2017-10-09:
@carlonb can you provide config and stl so others can reproduce the behavior? It is hard to debug the issue with only a piece of gcode output.

carlonb, 2017-10-09:
Sorry, forgot it.
Here the files.
The object is a 10x10x8mm tower divided in 4 pieces 10x10x2mm each in order to print it with four extruders, note that in this my attempt (to learn it) and annexed gcode I used only two extruders, 1 and 2 (T0 and T1).
Thankyou
Carlo
[Colonna10x10-4colori_config.zip](https://github.com/alexrj/Slic3r/files/1368447/Colonna10x10-4colori_config.zip)
[STL_Colonna10x10-four-colors.zip](https://github.com/alexrj/Slic3r/files/1368451/STL_Colonna10x10-four-colors.zip)

platsch, 2017-10-13:
Your config (slic3r.ini) configures only one extruder. But anyway, I can reproduce the issue after adding a second extruder. Didn't yet find the time to track down the problem, but seems to be important...

platsch, 2017-10-13:
OK, what happens is following: your first part (blue) ends exactly at z=2.0mm, the second part (red) starts at exactly z=2.0mm. With your settings, the layers will be printed at 1.85 and 2.15. slice_z is in the middle of each layer, so the intersection is computed at exactly 2.0mm, resulting in 2 polygons at the same layer, one for each volume.
Quickfix: slightly adjust the layer height or first layer height (e.g. set `First layer height` to 0.3501) to avoid hitting the double face or create an stl with a small gap.

Unsure if this should be fixed or somehow avoided in Slic3r itself...

carlonb, 2017-10-13:
Hi platsch,
OK, I see and will try it.
Thanks

If you have a time, there is also another strange behavior: 
If you enable the skirt, this will be divided in two parts with an intermediate tool change, I think this is not necessary.
Bye
Carlo

platsch, 2017-10-13:
If you enable skirt with more than one extruder, Slic3r distributes the loops over all extruders to ensure every extruder is primed correctly (which is the idea of using a skirt).

carlonb, 2017-10-13:
About skirt, so if I have 4 extruder I will have 4 tool changes ? This seems excessive. We do not need this as in tool change I (we) do a 30/40mm purge of the new filament.

platsch, 2017-10-13:
What if you have an object where both colors occur at the same height? e.g. place your red block next to the blue block. In that case you want a toolchange in every layer.

platsch, 2017-10-13:
just set the number of skirt loops to 1 and only your first extruder is used.

carlonb, 2017-10-13:
I understand, you are right about multiple tool change on the same layer, but not needed (i think) in skirt.
Thanks for the clarify
Bye
Carlo

carlonb, 2017-10-13:
> OK, what happens is following: your first part (blue) ends exactly at z=2.0mm, the second part (red) starts at exactly z=2.0mm. With your settings, the layers will be printed at 1.85 and 2.15. [...]
Quickfix: slightly adjust the layer height or first layer height (e.g. set First layer height to 0.3501) to avoid hitting the double face or create an stl with a small gap.
Unsure if this should be fixed or somehow avoided in Slic3r itself...

Hi platsch,
About this, sorry if this reply is so stupid, but as I wrote before and I don't know if you already read it, 
.... if you do a thing like:
If the last layer height is = to the next, skip this layer and go to the next height
????
Bye
Carlo

platsch, 2017-10-13:
At which point would you do this check? From Slic3rs perspective, there is no "next layer" in your case, it is a region in the same layer with a different color/filament/extruder. It can be seen as "you are slicing an object with overlapping volumes" and therefore Slic3r correctly generates overlapping gcodes.
I admit that an infinitely small overlap is exactly the one case where you probably don't want this behavior, but the solution would probably be located in the `slice` core-method and to me it is not very obvious without further exploration whether this is a good idea or not an how to exactly implement it.

carlonb, 2017-10-13:
OK platsch,
I see it's not so simple.
I will try your suggestion changing the first layer height. My concern is that every time I do a slice I need take a look at the sliced g-code in order to verify the presence of this overlapped layer.
May be once I change the first layer height from 0.35 to 0.3501 (for example) of one object, may be all my sliced objects will be correct if I use always the same layers height.
Thankyou very much.
I will try and report.
Carlo

platsch, 2017-10-13:
I agree that this solution is not very satisfying. @alexrj  @lordofhyphens any comments? have you already discussed this at some point?

lordofhyphens, 2017-10-14:
Currently I am leaning towards GIGO and those two volumes should not be intersecting.

carlonb, 2017-10-14:
@lordofhyphens 
Hi, for my understanding, what is GIGO ?
Thankyou
Carlo

lordofhyphens, 2017-10-14:
Garbage in; garbage out :)

Means that if the input is weird, getting weird output is expected.

While I am still open to software changes, I would prefer to see a test
case that doesn't have overlapping volumes that does something weird.

<https://github.com/alexrj/Slic3r/issues/4149#issuecomment-336639208>, or mute
the thread

carlonb, 2017-10-14:
Ok, yes I know garbage in, garbage out...
So please, are you saying that in my example you see a sort of my bad input/configuration or it's a sclic3r bug?
If it's my bad slic3r configuration please give me an hint.

lordofhyphens, 2017-10-14:
@platsch indicated that there are two different overlapping volumes (assigned to different extruders). Slic3r doesn't have a way to merge the two together (nor a way of knowing that it is appropriate to do so), so Slic3r generates gcode for both volumes in the same area.

lordofhyphens, 2017-10-14:
Also the skirt purge should only happen if there is a print move in that layer.

carlonb, 2017-10-15:
Layer overlapping issue:
I do not know nothing about slicer engine nor the whole s/w about this kind of programs, but why with the same layer config (first layer heitgh=0.35mm all the other=0.30mm)  Cura slicer do not overlap  this at height 0.2150mm?
Skirt purge:
Ok, this is a Slic3r philosophy, anyway (Cura do not do this) I still think that this will be left at operator decision to purge the nozzle, I do not need it because during tool change I purge the new tool.
What happen if my printer setup has 4 extruders but I will use only 2? Will purge 4 or 2 in skirt?

Sorry for the mention of Cura, I still use Slic3r cause it's my preferred. :)
Thanks
Carlo

lordofhyphens, 2017-10-15:
If you only use 2 during the print it should only purge 2. Also if you
disable skirt it should not do it at all.

platsch, 2017-10-16:
I don't agree with brushing this as GIGO. Defining the border of an object is often ambiguous. Having one object ending at 2.5 and the next one starting at exactly this height can be a valid approach. There is simply no good approach to represent a plane of zero thickness. The other option would be adding a small gap between the objects. If you are unlucky, slice_z happens to hit that gap and the layer is entirely omitted.
That having said, I still don't have a good idea how to solve or mitigate the problem...

carlonb, 2017-10-23:
I thought a while, so I'm asking:
1) Why this overlapping issue only on one layer and not on others since the Z increment is constant ?
2) Why is this issue only if there is a tool change ?
3) If only one extruder is used, the overlapping layer does not happen while the "ambiguous" Z increment still exists.
Thanks.