slic3r/Slic3r #4150: [Feature Request] Sequential printing of parts in multi-material objects

syl405, 2017-10-12:
### Version 
1.3.0-dev

### Operating system
Lubuntu 17.04 (64-bit, in a VM)

### Overview
I am trying to automate G-code generation using Slic3r for a multi-material, multi-process workflow involving embedded 3D printing (process described [here](https://lewisgroup.seas.harvard.edu/files/lewisgroup/files/embedded_3d_printing_of_strain_sensors_within_highlystretchable_elastomers.pdf)). This is part of an emerging set of related techniques that essentially use either [curable](http://www.nature.com/doifinder/10.1038/nature19100) or [soluble](http://advances.sciencemag.org/cgi/doi/10.1126/sciadv.1500758) gel baths as support material to enable deposition of various types of inks in 3D space. This has been used to fabricate things ranging from soft sensors and robots to complex biological structures.

The specific feature I would like to discuss in this issue is Slic3r's sequential printing behavior with respect to AMF files with multiple objects/multi-part objects.

### Current Behavior
I am starting with [an AMF file](https://github.com/alexrj/Slic3r/files/1378757/shell_2_inks.zip) with multiple objects. When I import the AMF, I choose the option to consider it as a single object with multiple parts in order to preserve the relative Z positioning of the various volumes (i.e. the text "floating" in the interior volume described by the box). In the image below, the green box is intended to be printed using FFF, while the "floating" text in red and yellow are intended to be embed-printed using a syringe extruder and supported by the gel reservoir.
[image]

Since the embed-printing process depends on having a poured gel reservoir (so that deposited parts are supported), these volumes have to be printed sequentially. I.e., the FFF mold will have to be completed, then the gel reservoir poured (let's just assume this is done manually for now), then each "floating" volume printed in what Slic3r thinks is mid-air. This is currently not possible because sequential printing takes place at the object level instead of the object volume level, so Slic3r attempts to print the floating text with within-layer toolchanges, resulting in layers like this:
[image]

### Target Behavior
My goal is to be able to import a multi-material AMF file comprising one or more multi-volume objects, and be able to print object volumes sequentially in a user-specified order. In the case of the attached AMF file, I would like to generate G-code to automatically carry out multiple steps in an embedded 3D printing workflow with the appropriate tool changes between steps:
- Build a mold using a FFF extruder
- Pour the correct amount of reservoir material (gel) into the mold
- Embed-print inside the volume of the gel reservoir using a ink 1
- Embed-print inside the volume of the gel reservoir using a ink 2

### Suggestion
I can see two ways to modify Slic3r's behavior that would achieve what I want:
1. Select "no" to considering the AMF a multi-volume object during import, and modify the "drop to z" behavior associated with independent objects. This would enable sequential printing like I intended. However, as pointed out in #1063 this is a inferior solution because of the blurring of semantics surrounding objects and object volumes. Even with built-in support like a gel reservoir, some part (in this case the mold) of each object should still touch the printbed/raft.
2. A better solution is to extend sequential printing to also work _within_ objects such that individual part volumes are finished before moving on to the next. For this to work in the correct sequence, the merging and ordering capabilities described in #3266 will also have to be implemented.

What do people think? I would love to contribute to implementing this, as well as #3266. As this would be my first time contributing to Slic3r, pointers on where to start looking (files, classes, functions) would be much appreciated!

### AMF file
[shell_2_inks.zip](https://github.com/alexrj/Slic3r/files/1378735/shell_2_inks.zip)

lordofhyphens, 2017-10-13:
- http://manual.slic3r.org/libslic3r-doc/ has libslic3r's own documentation.
- Print::validate() will probably need to be checked to ensure that you still obey height clearances.
- The main code that handles arranging print moves for multiple complete objects is in GCode.pm, line 219. It would be preferable to us if you replaced that implementation with a C++ one that would be accessed through GCode.cpp
- Model.cpp and Model.hpp in xs/src/libslic3r are worth looking over as well.

syl405, 2017-10-13:
@lordofhyphens Awesome, I will take a stab at it!

platsch, 2017-10-13:
Thumbs up for implementing #3266.
Regarding your request in general: it seems to be a very specific use case. Is there anything else you could achieve with sequential printing within a single object?
Personally, my first impression is that your first suggestions is a more general and clean solution. I don't fully agree with @alexrj argumentation in #1063. The primary use-case I see besides your application would be printing on top of something else, e.g. an older object one wants to extend or repair. Problem here would be registration (where exactly to put the existing object on the printbed).

So my question for consideration basically is: is it worth to introduce the extra complexity into slic3r or is this single purpose and should be solved locally?

syl405, 2017-10-13:
@platsch To clarify, the crux of my request is sequential printing on top of/within an older object. Sequential printing within a single object is simply my initial take on a good way to implement this. In fact, after doing some digging through Slic3r's innards, I am inclined to agree with you that my first suggestion is a cleaner solution -- nesting another layer of sequential printing behavior _within_ objects would introduce significant extra complexity. I can probably get the behavior I want by combining #4146 with a few tweaks to Print::validate().

Regarding the reasoning in #1063, I think the key difference here is that unlike with FFF, "floating" objects are in fact possible with embedded printing. The philosophical question, then, is whether it's worth extending Slic3r's capabilities to do embedded printing (as well as the repair/extend use case you point out), or whether it's unnecessary feature creep. I think in this case, using #4146 and the code I plan to write towards #3266, it would make Slic3r useful for people trying to do this stuff with minimal changes to current behavior.

lordofhyphens, 2017-10-13:
@syl405 to answer your IRC question, if I remember correctly PrintRegions pertain to modifier meshes.

syl405, 2017-10-14:
@lordofhyphens got it, thanks!