slic3r/Slic3r #4151: Z-Lift stepper hold Time-out while waiting for extruder temperature

Gnat666, 2017-10-14:
### Version
1.37.1-prusa3d-win64

_Use `About->About Slic3r` for release versions._

_Do not report Prusa3D Slic3r bugs here without confirming it is a problem on a development release of Slic3r, or your issue will be closed. *Only* use normal Slic3r version IDs._

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 7 Version	6.1.7601 Service Pack 1 Build 7601

### Behavior
Creality CR10 PLA extrusion, Bed set to 55, extruder at 195

- Select file to print, started the print.
- Unit waits until the bed reaches temp
- Unit then moves to 0/0/0.
- Unit then lifts the extruder a bit.
- Unit waits for the extruder head to reach temperature.
- At beginning of this wait it is noted that the Z-Lift stepper is locked, it can not be moved with fingers.
- *** Unfortunately after approx 1 minute, before the extruder reaches temperature, the Z-Lift stepper is released.
- *** confirmed not held because Z-Lift coupling can be turned with a finger
- *** release usually results in a gravity Z-Lift drop of a small amount while waiting for extruder temperature.
- *** once at temperature, the extruder travels to the start but now has lost its calibration on the Z-axis.
- *** Causing extruder to crash to bed and bad first layers.

Only way to bypass is to cancel & restarted the print immediately, this way the Unit is close to temperature and reaches temperature parameters before the unknown time-out kicks in.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Gnat666, 2017-10-14:
... why won't file attach ....
Anyway, first bits of gcode are;
; generated by Slic3r 1.37.1-prusa3d-win64 on 2017-10-14 at 20:01:37

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.40mm

M107
M190 S55 ; set bed temperature and wait for it to be reached
M104 S195 ; set temperature
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle

; Filament gcode

M109 S195 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
G1 Z0.400 F4800.000
G1 E-6.50000 F2400.00000
G92 E0
G1 X89.503 Y220.008 F4800.000
G1 E6.50000 F2400.00000
G1 F1800
G1 X90.681 Y219.652 E6.64614
G1 X91.889 Y219.418 E6.79229
G1 X93.114 Y219.307 E6.93843
G1 X175.287 Y216.767 E16.70231
G1 X182.363 Y216.762 E17.54274
G1 X183.592 Y216.825 E17.68889
G1 X184.817 Y217.014 E17.83605
G1 X186.016 Y217.327 E17.98322
G1 X187.177 Y217.760 E18.13039
G1 X188.242 Y218.285 E18.27142
G1 X189.258 Y218.915 E18.41348
G1 X191.361 Y220.653 E18.73739
G1 X192.366

Gnat666, 2017-10-14:
And Config

# generated by Slic3r 1.37.1-prusa3d-win64 on Sat Oct 14 21:18:25 2017
avoid_crossing_perimeters = 1
bed_shape = 0x0,300x0,300x300,0x300
bed_temperature = 55
before_layer_gcode = 
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_angle = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 40
brim_width = 0
clip_multipart_objects = 0
complete_objects = 0
cooling = 1
default_acceleration = 0
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
duplicate_distance = 6
elefant_foot_compensation = 0
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M104 S0 ; turn off temperature\nG28 X0  ; home X axis\nM84     ; disable motors\n
ensure_vertical_shell_thickness = 0
external_fill_pattern = rectilinear
external_perimeter_extrusion_width = 0
external_perimeter_speed = 50%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #FFFFFF
filament_cost = 16
filament_density = 0
filament_diameter = 1.75
filament_max_volumetric_speed = 0
filament_notes = ""
filament_settings_id = 
filament_soluble = 0
filament_type = PLA
fill_angle = 45
fill_density = 15%
fill_pattern = triangles
first_layer_acceleration = 0
first_layer_bed_temperature = 55
first_layer_extrusion_width = 200%
first_layer_height = 0.4
first_layer_speed = 30
first_layer_temperature = 195
gap_fill_speed = 20
gcode_comments = 0
gcode_flavor = reprap
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 25%
infill_speed = 60
interface_shells = 0
layer_gcode = 
layer_height = 0.2
max_fan_speed = 100
max_layer_height = 0
max_print_speed = 80
max_volumetric_extrusion_rate_slope_negative = 0
max_volumetric_extrusion_rate_slope_positive = 0
max_volumetric_speed = 0
min_fan_speed = 35
min_layer_height = 0.07
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
octoprint_apikey = 
octoprint_host = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 40
perimeters = 3
post_process = 
print_settings_id = 
printer_notes = 
printer_settings_id = 
raft_layers = 0
resolution = 0
retract_before_travel = 2
retract_before_wipe = 0%
retract_layer_change = 0
retract_length = 6.5
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
seam_position = aligned
serial_port = 
serial_speed = 250000
single_extruder_multi_material = 0
skirt_distance = 10
skirt_height = 1
skirts = 3
slowdown_below_layer_time = 5
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 20
spiral_vase = 0
standby_temperature_delta = -5
start_filament_gcode = "; Filament gcode\n"
start_gcode = G28 ; home all axes\nG1 Z5 F5000 ; lift nozzle\n
support_material = 0
support_material_angle = 0
support_material_buildplate_only = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_contact_loops = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 40
support_material_synchronize_layers = 0
support_material_threshold = 0
support_material_with_sheath = 1
support_material_xy_spacing = 50%
temperature = 195
thin_walls = 1
threads = 8
toolchange_gcode = 
top_infill_extrusion_width = 0
top_solid_infill_speed = 15
top_solid_layers = 3
travel_speed = 80
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 1
wipe = 0
wipe_tower = 0
wipe_tower_per_color_wipe = 15
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
xy_size_compensation = 0
z_offset = 0

lordofhyphens, 2017-10-14:
1. Looks like there are no explicit commands from Slic3r to turn off motora (m84). At one point Marlin did turn off z when it wasn't needed. 
2. You are reporting using prusa3d fork, which is not supported here. In the future you need to confirm on the main Slic3r or I will close the issue until you reproduce on main Slic3r  (preferably the newest dev build from dl.slic3r.org/dev)
2. Timeouts are governed by your firmware, not Slic3r. 
3. You should be able to attach zip files.
4. Add m109/m104 commands in start gcode to get more control. Wiki has more information. 

If you can find a cause-effect in the gcode output then I will reopen.

closed 2017-10-14 by lordofhyphens