slic3r/Slic3r #4154: Checkboxes do not show tooltips on Ubuntu

Zip-o-mat, 2017-10-17:
### Version
1.3.0-dev & 1.2.9

### Operating system type + version
Ubuntu 16.04 LTS

### Behavior
Checkboxes do not show tooltips on Ubuntu 16.04.
The tooltips are correctly shown on Windows 7.
When you click on the checkbox, the tooltip appears correctly.

The Checkbox tooltips seem to work correctly in a simple example:
[test.pl.zip](https://github.com/alexrj/Slic3r/files/1390440/test.pl.zip)

lordofhyphens, 2017-10-17:
Pictures, please? Gnome screenshot lets you take screencaps on a delay. Use alt-printscreen on windows.

Zip-o-mat, 2017-10-18:
I build the version from the git repo myself. I just tried the appimage build, it has the same problem.
The old 1.2.9 Build does not work either.

When you hover the mouse over the checkbox the tooltip does not appear, regardless how long you wait.
Tooltips on the textfields work as intended.
[image]

When you click the checkbox and stay there with the mouse, the tootip appears until you leave the checkbox. When you enter the checkbox again the tooltip will not appear.
[image]

It would be interesting if this works on MacOS, but I cant test it.

lordofhyphens, 2017-10-19:
This is showing up on Debian Jessie as well :( 
It does show the tooltip for a hover after a click. I suspect it has to do with getting/losing focus.

paulfertser, 2017-11-08:
Another datapoint: I'm seeing the same issue on Gentoo with XMonad. When I hover over a checkbox I see it highlighted but the tooltip doesn't appear. When I switch to another desktop and back, the tooltip appears. This affects checkboxes but not editboxes.
dev-perl/Wx version 0.992.800, wxGTK version 3.0.2.0-r2.

andy5995, 2019-01-09:
> This is showing up on Debian Jessie as well :(
> It does show the tooltip for a hover after a click. I suspect it has to do with getting/losing focus.

On Stretch, I can't get the (checkbox) tooltip to appear at all. I click in the box, hover, wait, click, hover, wait.

This happened on the 1.29 distribution in the stretch repo, and now I'm using a 1.30 self-build.

I used the build instructions [here](https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux#build-slic3r) and installed the perl modules with cpan.

lordofhyphens, 2019-01-09:
@andy5995 appimage?

lordofhyphens, 2019-01-09:
May be related: https://forums.wxwidgets.org/viewtopic.php?t=19044
It's looking like it may be a wxGTK bug.

lordofhyphens, 2019-01-09:
I can't do anything about the wxWidgets version or its usage at this time.