slic3r/Slic3r #4159: [Feature Request] Configurable (rectilinear) infill angle stepping

JohannStudanski, 2017-10-19:
### Version
1.3.0-dev

### Operating system type + version
Windows 10 CA / MacOS X Sierra

### Behavior
Currently, infill has a defined stepping between layers, either 0 (e.g. honeycomb) or 90 degrees (e.g. rectilinear).

It would be nice if this angle were configurable.

This would give e.g. 3*n spoked circular designs more (consistent) strength (set angle advance to 60° for a 3 spoke design for example), think hexacopter or similar things.

lordofhyphens, 2017-10-30:
The minor trick here is getting a configuration option down to the infill configuration, but this is doable.

JohannStudanski, 2017-10-30:
Hmm, I've used this with the Craftware slicer a few times, and didn't get bad results, but then again, I did run relatively low layer heights and pretty fat extrusion for my infill. So in other cases it might become an issue, yeah.

Thanks for taking the time to look into it, in any case!

drewtm, 2018-09-14:
This has been posted for a while but I wanted to chime in that I also want this feature. Two particularly applicable cases are when printing thin rings and when printing large, full-perimeter-support bridges. (An upside-down cup would have both of these features.) With the two perpendicular infill directions on a thin ring, one ends up with quite different structure and properties in the four "cardinal direction" areas of the ring than in the four 'diagonal' areas. In my attached image, the two highlighted areas have quite different structural properties, as well a possibility of quite different levels of adhesion between the perimeters and the infill.

Does anyone have some hints on one or more relevant files to start looking at if we were to try to implement the feature?

[image]

lordofhyphens, 2018-09-14:
The fill patterns themselves are generated in FillRectilinear.cpp.

I would suggest also looking at PrintGcode.cpp.

lordofhyphens, 2018-09-14:
Also, Print/GCode.pm (PrintGCode.cpp is the reimplementation of Print/GCode.pm and will be used going forward).

lordofhyphens, 2018-09-14:
I do highly suggest also looking at the tests I've written in src/test/libslic3r; I am much happier with pull requests that have tests to cover their own functionality with them.