slic3r/Slic3r #4160: Find and integrate decent logging tool into libslic3r

lordofhyphens, 2017-10-20:
Currently, all of the libslic3r debugging messages are gated behind an `#ifdef` and consist of simple printf statements.

It would be very convenient to incorporate a more functional logging library system, preferably with some ability to (at runtime) enable/disable different output streams. 
Something that acts like iostreams would be a big plus (ease of use) and the ability to redirect to files.

Boost.Log does look promising.

bubnikv, 2017-10-20:
Boost::log works very well for the Prusa3D fork. It helps with debugging and profiling.

lordofhyphens, 2017-10-23:
I had done a one-off when I was working on the conditional gcode feature, couldn't get it to compile in the bit I was messing with it. :/

When I find some time I will probably mess with it more. Boost::log was certainly convenient as we are already using other things in Boost.

lordofhyphens, 2017-10-24:
When I tried it again, it decided to work. :/

Something I do definitely want is ability to control the logging output at runtime. Probably with a singleton Slic3r::Log instance, although that would be just there to provide a way to change/set the logging sink configuration (at least the level) at runtime.

bubnikv, 2017-10-24:
I believe the logging in the prusa fork is controlled with an env value
SLIC3R_LOGLEVEL. The C++ interface is published into perl, so the logging
behavior is unified.