slic3r/Slic3r #4162: [Feature request] "External perimeters first" setting separate for first layer

jkavalik, 2017-10-20:
Hello and thank you for this great piece of software.

I configure my prints for internal perimeters to go first but I noticed on my last few current items that it would be nice to have the external perimeters go first for the first layer in some cases.

Specifically I am printing snowflakes (PLA and PETG) and the first layer starts with the small circles or loops which do not hold in place properly, because as the head returns in the loop, it sometimes catches the entire piece and carries it away before it cools down enough to stick properly, but it is not a problem for the longer external perimeters.

lordofhyphens, 2017-10-21:
Can you check real quick and see if you can do it with a modifier mesh?

Make a slab modifier your first layer thickness and check to see what
options are there.

jkavalik, 2017-10-21:
Tried it and yes, "External perimeters first" is available under "Layers and perimeters" on the modifier and seems to work according to gcode.ws rendering of the generated code, thank you.

closed 2017-10-21 by lordofhyphens