slic3r/Slic3r #4165: Feature Request - Nozzle position in 'Preview' and "Layers" window

new-yen, 2017-10-22:
### Version
Slice3r
Version 1.36.2

### Operating system type + version
Windows

### Behavior
It would be great to be able to see the nozzle at the beginning and the end of the layer.
This would help to evaluate the smoothest outer surface in regards of the various 'seam position' to get the best surface quality.
The beginning and ending positions could just have a colorful marker.

[image]

thanks and best regards
david

platsch, 2017-10-23:
Ordering of the extrusion paths and non-extrusion movements are only generated for gcode export. So this would require some additional processing each time the preview is rendered.
Despite of that, I'd like to have this feature.

dartrax, 2018-01-22:
+1 for that.