slic3r/Slic3r #4166: Feature request: STL for build volume definition.

rewolff, 2017-10-22:
The build volume is currently specified as a simple 2D form (circle or rectangle) extruded through a height. 

Many FDM machines do not exactly fit  this bill. For example, my I2 like printer has a rectangular platform that has a few screws in the corners. So now we define a build volume as the square that fits between the screws and we're done. But in fact, the printer can reach the screws. Just not at the plate level. 5mm above that it's fine to print there. No support allowed, but something that extends into such a corner is fine. 

At the top end of the build volume, my X axis runs onto stuff up there, so I can't really use the full X range above a certain height anymore. 

Similarly, my delta printer now uses a circle that must fit inside the screws that hold down the build plate. But a biger circle with three small cilinders removed at the screw positions would extend my build capabilities. Also, I suspect that even if the diameter of the build platform is 22cm *radius 11cm), it can reach further than 11cm from the center of the plate. So no building on the plate outside 11cm, but higher up that would be posible. (again, no support structures allowed, but overhanging pieces of hte object is possible). 

And again at the top. The build volume of a delta printer is not limited by the flat top of a cylinder, but by the intersection of three spheres. To be safe you'd declare the cylinder with the flat top that fits entirely inside the real build volume as the build volume, but some high objects that extend above that can be printed as long as the object stays in the middle and does not come close to the towers. 

Now you can design a language to specify more complex forms than just the box and cylinder. Or we could use STL that already exists and is already supported in the program. 

For those who know openscad, the current build volume of my I2 is: 
 cube ([200,200,110]);
while it could be: 
  difference () { 
    cube ([220,220, 110]), 
    translate ([ 10, 10,0]) cylinder (r=15, h=5);
    translate ([210, 10,0]) cylinder (r=15, h=5);
    translate ([210,210,0]) cylinder (r=15, h=5);
    translate ([ 10,210,0]) cylinder (r=15, h=5);
 }

I've put some unnecessary "keepout" zone around the screws to make it simpler. 
I'd have to move the X and Y endstops to prevent damaging them when going below current X=0 or Y=0. 
The firmware would have to learn not to place the head at 0,0 when zero-ing the Z as the screw is there. 

But without those changes, I could already print about 1cm more on the other side.

lordofhyphens, 2017-10-23:
So what does the configuration look like?

lordofhyphens, 2017-10-23:
Right now, the configuration entry for bed shape looks something like: 
````
bed_shape = -147.5x-147.5,147.5x-147.5,147.5x147.5,-147.5x147.5
````
That's for a simple, square bed. Once you go 3D, it gets a lot more complicated.

We also don't display the build volume.

rewolff, 2017-10-23:
So what you're saying is that the current definition defines a polygon that is the shape of the printable bed. So my "slightly bigger" printable bed would simply exclude the corners with the screws. 

But that still won't allow me to print above those screws once there is enough clearance, right? Or I could  specify the bed including the screws and then remember to not place objects near the corners to prevent the screws from damaging the print head. This works well until one day you print something that needs to go close to a corner and also requires a big brim. 

If you allow an STL to specify the print bed, you get a lot of flexibility in one go. The bed shape is suddenly fully programmable. You just need to intersect the STL with a plane say 0.1mm from the object's bottom. Now this is a quite complicated 3D operation, but I believe the code to do this can be found somewhere in some opensource program. A slicer perhaps?

after rereading your question: currently the build volume is apparently defined by: 
# define build volume:
bed_shape = -147.5x-147.5,147.5x-147.5,147.5x147.5,-147.5x147.5
build_height = 200

and this should become: 
build_volume = prusa_i3_buildvolume.stl

For an implementation trajectory, we could extract the footprint and height from the STL and just use that in the beginning. Or we could start using the 3D build volume and build that from the bed_shape, build height for compatibility with older configuration files.