slic3r/Slic3r #4172: Bridge speed and flow ratio is used on incorrect solid infill layers.

TheBoojah, 2017-10-26:
### Version
1.2.9 and latest dev (oct 23)

### Operating system type + version
Win 10 latest patch

### Behavior
Bridge speed and flow ratio is used on incorrect solid infill layers.
Bridge should be use on the lowest top-surface solid layer, i.e. the first layer on top of the infill.
What I am getting is all solid layers except the very top and bottom ones use bridge speed and flow. So for 3 bottom 3 top, the first is ok, then 2 bridge layers - infill - 2 bridge and the final top layer.

To illustrate I take all the default settings, except changing bridge flow ratio to 0.1 to make it more visible:
[image]
The solid infill lines are very thin, indicating that bridge settings were used. 
This is just a 20x20x20 cube, and I believe there is no other settings changed in Slic3r - everything is default.

If this is not easily reproducible please let me know what additional info I can provide.
Thanks

lordofhyphens, 2017-10-26:
Please zip that model and an export config (Ctrl-E) and attach it to the issue (drag and drop)

TheBoojah, 2017-10-27:
I updated the issue with the requested files.

lordofhyphens, 2017-10-28:
As far as I am aware, the first "solid" infill over sparse infill is bridge settings by design (as that's really what it is, a bridge).

lordofhyphens, 2017-10-28:
It appears that all internal solid infill is being flagged as bridges, beyond the first layer above sparse infill.

It's definitely wrong for bottom layers that aren't above a void.

lordofhyphens, 2017-10-28:
Other notes: 
At 100% infill bridge infill is not used at all, so there's a special case going on here.

If you wanted to have a workaround for a simple case you can set the main mesh to 100% infill and make a modifier mesh that is scaled down slightly and set to the real infill density. Slic3r does not seem to pay attention to infill differences for modifier meshes for the purposes of internal bridge infill.

lordofhyphens, 2017-10-28:
I am pretty sure the error is in the surface type detection. Top and bottom surfaces are easy to find, so those are classified correctly. 

The associated code should be in Layer::detect_surfaces_type().

It's actually applied around line 205 of LayerRegionFill.cpp where the flow is calculated.

TheBoojah, 2017-10-28:
Thanks for looking into this bug!

The way I detected this is playing with the bridge settings, at 0.8 bridge flow ratio the thinner lines starts to get really noticeable in the preview. The real killer is however when it does 10mm/s on the second solid layer of a larger part. So I can se why nobody has reported this to far, as when you have default or "light" bridge settings you never notice a difference. On challenging parts with some more extreme bridge settings can really mess things up, a high bridge fan setting on a large second solid layer is not so great for bed adhesion for example.

Thanks for looking for a workaround. It won't work for my current model so another way is just to use close to default bridge settings and hope the bridges come out ok anyway.
 
I tested Slic3r 1.1.2 which worked correctly, only on the first solid layer over sparse infill is bridge settings used.

lordofhyphens, 2017-10-28:
Yeah, this is a regression from when the logic was ported.

wirlybird, 2017-12-24:
Hey guys, came across your posts searching for a solution to a similar issue.  This pic is the first layer over a support.  It is being treated as a bridge and making the extrusions too thin so they mainly do not attach to anything and when you remove the supports you have a stringy mess.
Increasing the flow percent just pushes the extrusions further apart so the gaps stay the same and it doesn't solve anything.
Now, I can understand why it may think it needs bridging here but it just makes a mess.

If I run the same part in S3D it builds a solid layer over the support and it is all good.

Would love to find a solution!  Even just being able to completely disable bridging would be good.

[image]

[image]

lordofhyphens, 2017-12-24:
It *is* a bridge. A bridge is any extrusion that doesn't have something
underneath it. In these case, things like layer height and width are
meaningless.

wirlybird, 2017-12-24:
Thanks, I understand it being a "bridge".  However, in a circumstance like this it creates an interface layer like in my example.
I looked at the same part in S3D and it also treats it as a bridge but the extrusions are tight together so it makes a good solid layer.  No saying one slicer is better than another!
Is there anything we can do to improve this first layer on top of supports?  Can we force it to a standard infill and perimeter layer?  Is there a way to even  just get standard extrusion widths?
Slic3r does a great job on this particular part of mine except for this one issue and is why I am using it.

I appreciate any ideas and happy holidays!!

lordofhyphens, 2017-12-24:
Width here is your nozzle diameter. No more, no less. You could tune the
placement to be more tightly packed but then we run the extra risk of
destroying the bridge with nozzle drag.

For this specific model, you could add a modifier mesh to set infill to
100% below the problematic area.

wirlybird, 2017-12-24:
I understand and thanks for the info.
I'll see if I can figure out how to do a modifier mesh!  Sounds like something I need to know!

wirlybird, 2017-12-24:
I found a tutorial to do the modifier mesh and I can get it to change the "infill" but I still can't seem to get it to affect the first layer over the support!!
Interesting all the same.

lordofhyphens, 2017-12-24:
Right, first layer over the support. I was thinking of something else.

You could set the interface spacing to 0, which might help a little. Or see
if you can change the bridge flow rate % at your modifier mesh (I do not
remember if this is possible).

wirlybird, 2017-12-25:
Thanks.  I tried messing with the bridge flow rate but it doesn't seem to make a difference.
I'll see about the interface spacing.

bubnikv, 2017-12-25:
This may be of interest to you.
https://github.com/prusa3d/Slic3r/issues/102
https://github.com/prusa3d/Slic3r/issues/612

Namely, there is a workaround to force Slic3r to squish the bridging
threads against the support interface layer.

wirlybird, 2017-12-25:
Thanks for the info.  Interesting reading.