slic3r/Slic3r #4174: Low brim to perimeter bonding

TheBoojah, 2017-10-28:
### Version
1.3.0 dev oct 23

### Operating system type + version
Win 10 64bit latest patch

### Behavior
The line spacing between the bottom outer perimiter and the innermost brim line is too wide. This is especially noticeable if a print is done with z-height adjustment so that the printed bottom layer is exactly or slightly taller than the set first layer height (for printing with high-stick build plate material for example). What happens is that the lines of the brim and the part adhere well to each other, but the brim is not attached at all or weakly to the part - rendering the brim useless.

1.3.0 dev:
[image]
Note the gap between the green brim lines and the part.

Correct results are obtained on 1.2.9:
[image]

#### STL/Config (.ZIP) where problem occurs
[Low brim to perimeter bonding.zip](https://github.com/alexrj/Slic3r/files/1424143/Low.brim.to.perimeter.bonding.zip)

lordofhyphens, 2017-10-29:
I'm thinking it's fp math somewhere. 

If you want to play around with the brim distance, see line 863 in Print.cpp. 

It shifts the loops in by half a scaled_width, so vary the value between 0 and -1. Positive values would put it further away.

lordofhyphens, 2017-10-29:
[image]

That's with a -0.55 modifier, which would provide a more firm connection. It'd probably do as a workaround for your immediate needs.

lordofhyphens, 2017-10-30:
I've pushed a workaround.

paulfertser, 2017-11-08:
#2230 seems to be related here. Apparently the connection was nice and strong before af46db320f820cc68e692cccf2f83890f4ee1596 and in this ticket the workaround is not really a revert of it. Was this considered? I would expect strong brim connection to be the default behaviour and those who need to separate it more (like the #2230 reporter) should get a configuration option.
(just had one ABS part lifted off bed together with internal brim, with outer brim still there on the glass)

lordofhyphens, 2018-07-11:
I suspect we can't get around a config option now for brims distance (interior and exterior). I was really hoping to avoid it.

lordofhyphens, 2018-07-11:
For people wanting to take a swing at this, I would recommend using a drop-down box in the UI with a few sensible defaults (the previous value, the current value, etc) and allowing the user to set it manually. If you go with a flat mm separation you'd need to back-calculate the separation. I'd suggest using % overlap (similar to the perimeter/infill overlap config option). 

af46db3 will guide you to where you would need to modify to make the relevant changes (this is in % of perimeter width if memory serves).

Tinchus2009, 2018-07-11:
well. Im always in the KIS side of solutions. But this issue should have never arised. The hole purpose of a brim is to be there, to give stronger adhesion to the object. If you weaken the brim... then why you want one? Anyway, i can just help with printing observations.

lordofhyphens, 2018-07-11:
@Tinchus2009 the reason it was weakened is because someone else complained it was bonding too well and required a lot of cleanup in their use case; and for them and their materials it did good enough with the weakened brim.

Welcome to my world: where users have conflicting needs.

Tinchus2009, 2018-07-11:
oh... yes I understand you and  im not sure if I want to be welcome in that world lol
But... haw hard can be to remove a 1 layer brim... just a utility knife  lol.
These features are "sensitive", forumstalks about this issue. We ahve coders here helping, we have "testers" like me helping too. But for each one of us, there are may be hunderd or thounds of people who just wont make a comment here just make a bad critic about the software. This issue is one of them. 
I guess there is no point now in a debate since the thing has been modified, but to me is kind of simple: A brim a brim, is is suposed to give a strong adhesion to the bed for object that eed it... if the bond to the object is weak... how it is suposed to give a strong adhesion???? it is just logic.
It is sad, I just cant use slic3r for abs now :( And for the record; when this setting was ok, removeing the brim was just a matter of useing a knife and remove it in no more than 30 secs from the model

lordofhyphens, 2018-07-11:
@Tinchus2009 which is why I think we need a configuration setting now. I didn't want to originally (because more one-off options are  confusing for users). 

I've left hints for someone to provide this as a PR.