slic3r/Slic3r #4176: Issue with sequential printing

gyrex, 2017-10-31:
### Version
1.3.0_dev

### Operating system type + version
MacOS 10.13

### Behavior
I'm attempting to run a sequential print. I've selected the "Complete individual objects" and defined the radius of my extruder. During the preview of the print, Slic3r is showing a normal parallel print.

_Is this a new feature request?_
No

I've included screenshots of the issue.

[image]
[image]

lordofhyphens, 2017-10-31:
Slic3r's preview is showing the layers.

What were you looking for from it? How would it be controlled?

gyrex, 2017-10-31:
I expected it to finish printing one part completely, then once it finishes printing that part, it starts printing the next part in sequence. The screenshot shows it printing a layer on one part, then moving to the next part and printing a layer on that part.

lordofhyphens, 2017-11-01:
There's no travel moves shown in the preview, and the slider units are in mm along the Z axis.

So how would you expect the preview to work as a gcode print preview (and not a layers preview)?

gyrex, 2017-11-01:
I've mostly used Simplify3D in the past so I'm using that application as a comparison, but there's some slicing algorithms which I think are better in Slic3r, particularly thin walls. In Simplify 3D, the preview shows the print process for each object in a sequential pattern. ie. it shows each layer being printed in object 1, when that's finished, it shows subsequent layers being printed on the next object. ie. exactly how it would happen on the build platform during the printing process. Here's some screenshots of S3D's preview which makes sense to me.

[image]
[image]

Are you saying that Slic3r will print the objects in sequential fashion but that preview function in Slic3r just shows how the objects look as you move up the build platform as if the print had finished?

lordofhyphens, 2017-11-01:
Correct. You can, of course, confirm this by examining the gcode output
(or gcode.ws, etc).

I just used Sequential Printing a few days ago, so I know it works :)

<https://user-images.githubusercontent.com/9957205/32260691-9f959fd8-bf1d-11e7-9984-1695510c2948.png>

[image: screen shot 2017-11-01 at 3 55 20 pm]
<https://user-images.githubusercontent.com/9957205/32260692-9ff87fcc-bf1d-11e7-8b2a-13bd5bac03a3.png>

<https://github.com/alexrj/Slic3r/issues/4176#issuecomment-340987744>, or mute
the thread

gyrex, 2017-11-01:
Thanks for your prompt response! Could I turn this into a feature request then? It would make a lot more sense to adopt a preview similar to S3D to show the tool movements across parts, rather than simply showing a layer by layer view of all parts, rather than the sequential view. Fantastic work on the software BTW. There's still a lot of things that Slic3r does a lot better than S3D.

gyrex, 2017-11-01:
Here's a video of how the sequential printing preview looks in S3D. In my opinion, it's much more user friendly, intuitive and useful than the current implementation in Slic3r. It would be fantastic to have something similar. Thanks again for the wonderful work!

https://youtu.be/sY-ymdjPrJk