slic3r/Slic3r #4178: Custom Gcode UI Redesign

lordofhyphens, 2017-11-01:
### Version
1.3.0-dev

### Rationale
The current mechanism(s) for inserting custom GCode into a print produces a lot of large text boxes and multiple configuration options exposed in libslic3r. I think it is possible to fold all of the current custom GCode (and the common use case of before/after layer X) into a single UI element that the user can expand/collapse as they wish.

### Technical

Custom GCode has two logical parameters: _What_ to be inserted and _When_ to insert it.
* _What_ is simple enough; a single string containing the gcode to be inserted.
* _When_ is the reason we have 6+ separate configuration options for custom GCode. 
     * To be useful at a minimum, a gcode snippet should be assignable to a layer (either before or after), or one of the known fixed occasions (already supported by Slic3r). 
        * Pre-print gcode
        * Post-print gcode
        * Toolchange (pre)
        * Toolchange (post)
        * All layer (pre)
        * All layer (post)
        * Between objects

### UI

UI-wise, the Custom Gcode textbox element would become be replaced by an option to add more Custom GCode textboxes that also have configurable fields for the _When_ parameters. Probably a dropdown for "Before/After X" and a free-floating field that would accept a comma-separated list of numbers to apply the same field to multiple layer numbers. I also propose an "before/after Z" parameter (also taking a list of Z heights) that rounds down to the nearest layer. 

Generalizing the UI in this manner permits applying most of the custom GCode options on a per-filament and per-print basis, either as an override (replace) or addendum (append). 

_Is this a new feature request?_
Yes

### Caveats

* Backwards compatibility for existing parameters would need to remain.
* Need a good CLI interface for specifying the expanded parameters (perhaps enforcing some order to custom gcode parameters so that the _When_ parameters precede the _What_ parameter they affect). 

This feature request would implement #3092 and re-implement #4177.
Some of this functionality is duplicated in Conditional GCode, but in a more discover-able method.

goofdad, 2017-11-01:
@lordofhyphens ... thanks for the rabbithole! Some thoughts:

>Users should be able to assign multiple snippets to the same general When and specify an ordering to those (with a default order generated).

This feels overcomplicated. I'm not seeing a use-case for multiple "before-layer" snippets, or "tool-change" snippets and then adding an order parameter. How is that different than just using them in the same snippet?

> a gcode snippet should be assignable to a layer

Yes, but as I said in my PR (#4177) I philosophically like the separation between the printer owning the gcode to insert and the print owning the list of layers. I'm just contemplating about a way to have different tags trigger different gcode (perhaps using the conditional syntax)

> Backwards compatibility for existing parameters would need to remain.

Is there a mechanism for versioning of parameters / config files? Is such a thing anathema for some reason that I'm missing? It seems to me it would also help re: #3574 

I just want to make sure I'm going down the correct rabbit hole and not out shaving yaks :)

lordofhyphens, 2017-11-01:
1. Once users see something in the UI, they will try to use it like that.
It has happened before, it will happen again. There wouldn't necessarily be
something to stop you from assigning 3 snippets to the same place and
layer, so we handle it as best we can. From the perspective of the system
it would not be any different, but it does let us capture the case of two
before gcode that do things and hitting the same layer on both (you want to
do X on layers 2, 9, 17 and Y on 19, 50 and x and y on 17 layer).

2. We try very hard to not break existing configuration files. Usually it
involves converting them on first load and using the new stuff from then
on. The way the config system is, it is at least crash resistant in terms
of forward compatibility.

<https://github.com/alexrj/Slic3r/issues/4178#issuecomment-341117093>, or mute
the thread

goofdad, 2017-11-01:
Rather than a button that can add items ad-nauseam, what if I implemented it as a dropdown with the G-Code options available, including the following:
* Tagged Filament Start G-Code
* Tagged Filament End G-Code
* Tagged Layer G-Code (perhaps "Before Layer" and "After Layer")?

Then provided "tags" in the Filament for Filament G-Code and in Print for Layer G-Code.

Upon start I could default extant filament gcode to tags named after the filament, and then encode them in an "if [tag]" in the Filament G-Code.

I'll have to figure out:
* How to dynamically change the gcode box
* How to create tag placeholders on the fly from the tags
* How to upgrade existing filament gcode (how big a priority is this as 1.2.9 didn't support it yet)

lordofhyphens, 2017-11-02:
Personally I think the button to add snippets is better UI and easier for people to mash things together.

goofdad, 2017-11-02:
With the UI learning curve about the same for both, I think I'll mock them both up, submit them both, and we can play and see what rises to the top :)

goofdad, 2017-11-02:
LOL ... this is fun. I'm learning WxWidgets as I got. Fun Fun Fun. If you could point me to a way to hide / show items in an $optgroup that would be great. If not, I'll figure it out ...

lordofhyphens, 2017-11-03:
The wxwidgets Web documents should be fine for the most part.

I believe that there is a Show and Hide methods on all of them.

foreachthing, 2017-11-15:
Whatever you do, just make that textbox bigger and hide that mouse-over-tooltip; maybe put it in an (?) next to the textbox. Depending on where you enter the textbox (with your mouse) you can't read the gcode anymore.

rob-miller, 2018-04-02:
The ability to 'include' gcode files would be nice for this, e.g. my PETG filament settings vary per spool due to preferred melt temperature, but all have the same retract and pressure advance settings for  my printer's firmware.  So if I could [include "petg_settings.gcode"] or similar that would be useful?