slic3r/Slic3r #4180: Conditional G-code issue with curly braces

goofdad, 2017-11-03:
### Version
1.3.0 dev (add2895)

### Operating system type + version
ubuntu 17.04

### Behavior
Just acquired a printrbot Simple Pro. It has an ABL sensor. According to printrbot's website you set the bed offset with the following command:
M100.1 ({{g55z: 0.0}}) ;Z-probe offset
where 0.0 is the offset. Thus, I put the following in my start_gcode:
M100.1 ({{g55z: -4.0}}) ;Z-probe offset

I expected that line to be carried through to my gcode, but when the code generates, it ends up as:
M100.1 () ;Z-probe offset

_Is this a new feature request?_
no

#### STL/Config (.ZIP) where problem occurs
[Failing.zip](https://github.com/alexrj/Slic3r/files/1439900/Failing.zip)

goofdad, 2017-11-03:
Printrbot site with startgcode:
https://printrbot.zendesk.com/hc/en-us/articles/115000151583-Printing-via-USB-on-Cura-2-with-Your-Printrbot-Simple-Pro

According to the Conditional Gcode draft (https://github.com/alexrj/Slic3r/wiki/Conditional-Gcode-Syntax-Spec) this should be passed through, yes?

lordofhyphens, 2017-11-07:
@goofdad yes, you are correct. I derped up on the "failure" case. That should be passed through as-is.

lordofhyphens, 2017-11-07:
Fixed. You can also escape `{}` with a single `\`, like `\{343\}` and it'll come out as `{343}`

lordofhyphens, 2017-11-07:
Added a test as well to catch that case.

closed 2017-11-07 by lordofhyphens

goofdad, 2017-11-07:
Thanks. I was just looking at the code. Going to try it out tomorrow when I have time to play with code again.

lordofhyphens, 2017-11-07:
Side effect: Please don't use ascii characters 128 or 129 in your gcode, they'll be converted to { and }.

commit f7e6d53e (2017-11-07): Allow the use of escaped open/close brackets so that brackets may be emitted in the output (does not permit use of {} inside evaluations).

Fixes behavior to emit brackets on failure.

Fixes #4180