slic3r/Slic3r #4181: strange path for calibration model

alexxy, 2017-11-03:
Version git master of 02.11.2017

Seems like on some dual extruder models (in that case [calibration one](https://www.thingiverse.com/thing:1526735)) slic3r generates strange paths...

[image]
[image]

lordofhyphens, 2017-11-03:
Please include everything that was requested in the text you removed. Exported config, the version (provided instructions for getting the version id or use the download name).

Also suggest retesting on a recent download

alexxy, 2017-11-03:
it was recent build from git.
add2895f38df40f79ff06552a179a3a88f3bf5e7

[a.zip](https://github.com/alexrj/Slic3r/files/1442687/a.zip)

lordofhyphens, 2017-11-03:
@alexxy Ctrl-E export, not a bundle, please. The bundle gives every configuration you have (so it's not a lot of help to us to test with). 

Thank you for posting the commit hash.

alexxy, 2017-11-04:
[config.zip](https://github.com/alexrj/Slic3r/files/1443096/config.zip)