slic3r/Slic3r #4182: Model looks normal in 3D view on Plater, but misses some after slicing

jdevelop, 2017-11-04:
### Version

`add2895f38df40f79ff06552a179a3a88f3bf5e7`

### Operating system type + version

Arch Linux 4.13.11-1-ARCH

### Behavior

The model is sliced incorrectly,

[image]
[image]

The source file:
[print.obj.gz](https://github.com/alexrj/Slic3r/files/1442927/print.obj.gz)

lordofhyphens, 2017-11-04:
Would be nice to have the Ctrl-E config export.

lordofhyphens, 2017-11-04:
Works after a pass through Netfabb's online service.
[print_fixed.obj.gz](https://github.com/alexrj/Slic3r/files/1442946/print_fixed.obj.gz)

jdevelop, 2017-11-04:
The config file example.

Also I tried the same model in Cura 3.0.3 - it was able to slice it just fine, if that helps.

```
# generated by Slic3r 1.3.0-dev on 2017-11-04 13:47:07
adaptive_slicing = 0
adaptive_slicing_quality = 75%
avoid_crossing_perimeters = 0
bed_shape = -5x-5,115x-5,115x115,-5x115
bed_temperature = 30
before_layer_gcode = 
between_objects_gcode = 
bottom_infill_pattern = rectilinear
bottom_solid_layers = 4
bridge_acceleration = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 10
brim_connections_width = 0
brim_width = 0
compatible_printers = 
complete_objects = 0
cooling = 0
default_acceleration = 0
disable_fan_first_layers = 1
dont_support_bridges = 1
duplicate_distance = 6
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
end_gcode = M104 S0 ; turn off hotend/extruder heater\nM140 S0 ; turn off bed heater\nG91 ; Switch to use Relative Coordinates\nG1 E-2 F300 ; retract the filament a bit before lifting the nozzle to release some of the pressure\nG1 Z1 ; raise Z 1mm from current position\nG1 E-2 F300 ; retract filament even more\nG90 ; Switch back to using Absolute Coordinates\nG1 X20 ; move X axis close to tower but hopefully far enough to keep the fan from rattling\nG1 Y120 ; move bed forward for easier part removal\nM84 ; disable motors\nM107 ; turn off fan. Some say M107 doesn't work for them but if the hotend is cool enough it should work
external_perimeter_extrusion_width = 0.4
external_perimeter_speed = 15
external_perimeters_first = 1
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0.4
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #000000
filament_cost = 17
filament_density = 0
filament_diameter = 1.75
filament_max_volumetric_speed = 0
filament_notes = ""
filament_settings_id = 
fill_angle = 45
fill_density = 0%
fill_gaps = 1
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_bed_temperature = 40
first_layer_extrusion_width = 0.42
first_layer_height = 0.2625
first_layer_speed = 20
first_layer_temperature = 200
gap_fill_speed = 20
gcode_arcs = 0
gcode_comments = 1
gcode_flavor = reprap
has_heatbed = 1
host_type = octoprint
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0.4
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 25%
infill_speed = 20
interface_shells = 0
interior_brim_width = 0
layer_gcode = 
layer_height = 0.0875
match_horizontal_surfaces = 0
max_fan_speed = 99
max_layer_height = 0.30625
max_print_speed = 30
max_volumetric_speed = 0
min_fan_speed = 100
min_layer_height = 0.0875
min_print_speed = 10
min_skirt_length = 0
notes = 
nozzle_diameter = 0.4
only_retract_when_crossing_perimeters = 0
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 1
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0.4
perimeter_speed = 20
perimeters = 2
post_process = 
pressure_advance = 0
print_host = 10.0.0.28:5000
print_settings_id = 
printer_notes = 
printer_settings_id = 
raft_layers = 0
regions_overlap = 0
resolution = 0
retract_before_travel = 2
retract_layer_change = 0
retract_length = 5
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
seam_position = aligned
sequential_print_priority = 0
serial_port = 
serial_speed = 250000
shortcuts = support_material
skirt_distance = 5
skirt_height = 1
skirts = 1
slowdown_below_layer_time = 10
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0.4
solid_infill_speed = 15
spiral_vase = 0
standby_temperature_delta = -5
start_filament_gcode = "; Filament gcode\n"
start_gcode = M109 S[first_layer_temperature]\nG21 ;metric values\nG90 ; absolute positioning - this line was added in case Cura doesn't include it by default\nM82 ; \nG28 ; home all axes\nG92 E0 ; reset extrusion distance\nG1 Z5 F1200 ; raise nozzle 5mm\nG1 E12; Purge 12mm of filament \nG92 E0; reset extrusion distance
support_material = 0
support_material_angle = 0
support_material_buildplate_only = 0
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0.4
support_material_interface_extruder = 1
support_material_interface_extrusion_width = 0
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = pillars
support_material_spacing = 2.5
support_material_speed = 25
support_material_threshold = 60
temperature = 210
thin_walls = 1
threads = 8
toolchange_gcode = 
top_infill_extrusion_width = 0.4
top_infill_pattern = rectilinear
top_solid_infill_speed = 15
top_solid_layers = 4
travel_speed = 110
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
vibration_limit = 0
wipe = 1
xy_size_compensation = 0
z_offset = 0
z_steps_per_mm = 0
```

lordofhyphens, 2017-11-04:
Cura tries to do more with file repair than Slic3r does, so that doesn't
mean much.

As I noted, Slic3r does just fine after the model gets a pass through
netfabb. I will have to try to compare what Slic3r thinks the model looks
like with both versions. The fixed version is also larger iirc.

Currently I am leaning towards GIGO, but the input code needs to be looked
at to make sure there isn't a bug in it.

lordofhyphens, 2017-11-04:
[image]

Self-intersecting faces.

jdevelop, 2017-11-04:
Okay, that makes sense, it's the first time I tried to use FreeCAD to model something, and apparently I did something wrong. Thanks for the help1

jdevelop, 2017-11-04:
@lordofhyphens which tool did you use to figure that out?

closed 2017-11-04 by jdevelop

lordofhyphens, 2017-11-04:
Meshlab with the face view turned on showed a line going right through one of the pegs. 

It also has a "show self-intersecting faces" feature. 

In FreeCAD you want to do a boolean union on all of the components and extract the resulting mesh, IIRC.

jdevelop, 2017-11-04:
I regrouped the model a bit, so now it exports just fine. So at least I know - if something doesn't slice from the STL/OBJ exported by FreeCAD, and looks _okay_ after fixing the mesh - it's likely that the model itself is broken.

Thanks a lot again :)

lordofhyphens, 2017-11-04:
No problem. Happens a lot; solid 3D modeling is surprisingly finicky.