slic3r/Slic3r #4183: Combine infill every X layers produces unexpected results.

geofox784, 2017-11-06:
### Version
1.3 Dev 10-30

### Operating system type + version
Windows 7 64 bit

### Behavior
Combine infill every X layers doesn't work and produces unexpected results. 

* _Describe the problem_
When using 1.2.9 combine every 1 layer behaves as expected. Combine every 2 layers does nothing. It still creates an infill every layer. 
With 1.3 combine every 1 layer behaves as expected. Combine every 2 layers (or any value > 1) creates infill every 2 layers, but the infill "lines" are just as thin as what is created for combine every 1 layer. This results in gaps and a much weaker print. 

* _Steps needed to reproduce the problem_
See above. I did this with a 1 in cube STL file. 

* _Expected Results_
See above.

* _Actual Results_
See above.
From left to right
1. 1.2.9 every 1 layer
2. 1.2.9 every 2 layers (identical)
3. 1.3 every 1 layer (identical)
4. 1.3 every 2 layers (has gaps)

[image]
[image]

#### STL/Config (.ZIP) where problem occurs
Simply a 1 inch cube, or anything else.

platsch, 2017-11-10:
Can't reproduce on 1.3.0-dev. Printed an object with 50% infill, combine each 2 layers and it came out perfectly solid. Which infill pattern did you use? Can you provide a config so I can generate gcode from the same configuration?

geofox784, 2017-11-11:
I did this with 20% infill at .2mm layers. I just tried 50% and the problem was still there. [Here are some screenshot](https://imgur.com/a/RWpYE)s of the preview as another example. [Here is the config.](https://drive.google.com/file/d/1KWRGmgiQxJTewlPVf5zvoPoLTNe746rO/view?usp=sharing)

platsch, 2017-11-17:
Ok, you have infill type Rectilinear which is supposed to print lines in one direction per layer (or per x layers). Thats exactly what happens. You will always experience "gaps" with low infill density, because only one direction is filled alternating. For high infill values this results in solid layers.

I suppose the effect is just stronger if you combine infill layers because the gap between two aligned lines is bigger.

So I think it's not a bug.