slic3r/Slic3r #4184: 1.3.0 Release schedule?

platsch, 2017-11-06:
### Version
1.3.0

The 1.3.0 [milestone](https://github.com/alexrj/Slic3r/milestone/23) is almost complete and the website says Q4 2017 for release. I think a release is crucial for Slic3r, so much hast changed since 1.2.9. This is not a real Issue, but more of a question to better plan the next steps and where to put effort:

- Is there a planned release date?
- Besides the 5 points in the milestone, which other important tasks are still open? (e.g. website, documentation, ...)
- How can we help to get the release out?

lordofhyphens, 2017-11-07:
The actual release process is only doable by @alexrj (as he has the keys to the website), so it will happen when he gets the free time from whatever else he's doing right now.

* I had wanted it out when the milestone was marked. 
* Check https://github.com/alexrj/Slic3r-Manual/issues for open doc issues. The manual needs more work to update it for all the changes since 1.2.9. 
* Update docs and submit PRs for the open issues against 1.3.0.
    * The two real remaining issues are a nice-to-have feature request (button to apply overrides to your config) and a weird UI bug that I haven't been able to track down a cause for. Both could be deferred and a release could be pushed out.

stuartpb, 2018-01-11:
Seeing as how Q4 2017 has come and gone, what really needs to be done before 1.3.0 can ship?

lordofhyphens, 2018-01-11:
Last few items in the milestone need to be fixed or deferred.

Need to make a 1.3.0 version update commit (remove dev).

@alexrj needs to copy those builds to the appropriate area on the website and update links (@alexrj is the only person with the access rights to do that).

Work on the manual should be completed as well but that isn't a hard requirement.

stuartpb, 2018-01-11:
> Last few items in the milestone need to be fixed or deferred.

OK, well, looking at [all the remaining issues on the 1.3.0 milestone](https://github.com/alexrj/Slic3r/milestone/23):

- #4184 is this issue, which would be "fixed" by addressing all the other issues.
- #4096, on "updating the website", similarly, can really only be considered "resolved" by releasing 1.3.0 ("@alexrj needs to copy those builds to the appropriate area on the website and update links")
- #3834 looks like it was addressed, inasmuch as it should be allowed to block the next release, by the changes and discussion in #3912. (Further discussion on that issue can be left to affect the next release.)
- #4010 looks like it's been resolved, per the most recent comment [failing to reproduce with current builds](https://github.com/alexrj/Slic3r/issues/4010#issuecomment-342397498).
- This leaves the last issue, #4168, as the only issue that's *really* blocking the 1.3.0 release.

closed 2018-06-28 by lordofhyphens