slic3r/Slic3r #4185: Bug (small perimeters speed) + Option request

bigcatapult, 2017-11-09:
### Version
Slic3r-master.2017.11.07.1542.8ca6171.64bit

### Operating system type + version
Windows 10 
Version 1607
Build 14393.1593

### Behavior
What ever **Small perimeters speed** I'm set, it never goes higher then 17mm/s. (in GCode speed must be F2400, but it stays in range of F900-1000)
Pls add option to **disable Small perimeters speed.** (can't set manually constant flow with this feature enabled, because different speed(or wall width) ruins everything)

#### STL/Config (.ZIP) where problem occurs
[small_perimeter_speed_bug.zip](https://github.com/alexrj/Slic3r/files/1456332/small_perimeter_speed_bug.zip)

bigcatapult, 2017-11-09:
sorry. 
Problem is not in Small perimeter speed, but in slowing down, which affects not all speeds equally.

lordofhyphens, 2017-11-09:
Autospeed sets speed to maintain constant pressure.

If you are looking for constant flow  (gl!)  Fix all your widths to the
same and speeds to the same manually.

bigcatapult, 2017-11-09:
**Slow down if layer print time is below** does not affect external perimeters, so on bowden extruders it creates gaps and thinner walls if speed difference >150%.
I'm fixed previous link to discussion where is only topic starter admit adequate attention to this problem.

lordofhyphens, 2017-11-09:
Odd, it probably should affect external perimeters.

Area5142, 2018-01-02:
I expected the _cooling slow down_ to affect the hole layer, but read that it is intentionally turned off for external perimeters. I would like the "old" way to come back or at least the possibility to choose it.

Printing with TPU and Flex the _high speed_ external perimeters gives problems on small surface layers - the part tend to flex and move with the nozzle, distorting the shape and surface.
The slow down feature is needed to get the flexible filament a chance to cool down and the nozzle should move gently.
On larger areas and infill, it is fine to go faster and limit with the new feature (LA) and _max volumetric speed_.

With the "new" feature I have to slow down perimeter speed for the hole print to get a workable part.