slic3r/Slic3r #4186: [Patch] Compile error due to xs/src/admesh/portable_endian.h on FreeBSD

rmx-it, 2017-11-11:
### Version
1.2.9-1250-g8ca6171a

### Operating system type + version
11.1-STABLE FreeBSD 11.1-STABLE #2 r322118:

### Behavior
Compile warnings and errors in src/admesh/stlinit.c:30: 
...
use of undeclared identifier 'letoh32'
...

This is due the fact that xs/src/admesh/portable_endian.h redefines some macros wrong and unnecessarily.
The following patch fixes this:
````c
--- xs/src/admesh/portable_endian.h.orig	2017-11-11 17:01:40.937161000 +0100
+++ xs/src/admesh/portable_endian.h	2017-11-11 17:07:54.646935000 +0100
@@ -47,9 +47,9 @@
 #	define __PDP_ENDIAN    PDP_ENDIAN
 
-#elif defined(__OpenBSD__)
+#elif defined(__OpenBSD__) || defined(__FreeBSD__)
 
 #	include <sys/endian.h>
 
-#elif defined(__NetBSD__) || defined(__FreeBSD__) || defined(__DragonFly__)
+#elif defined(__NetBSD__) || defined(__DragonFly__)
 
 #	include <sys/endian.h>
````

lordofhyphens, 2017-11-11:
If you make a pull request I can accept it.

rmx-it, 2017-11-15:
Hi,
done: 
Fix compiling for FreeBSD #4188

udif, 2021-05-31:
@rmx-it , Can you please explain this patch to me?

From what I see above, you have moved the `defined(__FreeBSD__)` term one `#elif` branch to another, but they both enable reading the same `#include` file! (`<sys/endian.h>`).