slic3r/Slic3r #4191: Slic3r does not always retract when it should

VanessaE, 2017-11-17:
### Version

Git commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9
### Behavior

Load the attached .amf (it consists of two object - one main piece and one support piece).

Slice it, then look at the gcode output in something like gcode.ws.  Note how there is a travel move across a cutout in the front of the object from layers 23 through 32....  but there's no retraction happening here.  You can see the same thing if you actually print the object - the nozzle moves from one side of the gap to the other without invoking a retract, leaving the usual ooze/stringing across the gap.

#### STL/Config (.ZIP) where problem occurs
[config-20171117.ini.zip](https://github.com/alexrj/Slic3r/files/1482028/config-20171117.ini.zip)
[switch mount - power and fan.amf.zip](https://github.com/alexrj/Slic3r/files/1482040/switch.mount.-.power.and.fan.amf.zip)

lordofhyphens, 2018-04-10:
````
Moving to layer 22
POINT: 33.2078 43.8852 Needs retraction.
POINT: 33.2078 28.8745 Needs retraction.
POINT: 29.9216 28.4246 Needs retraction.
POINT: 24.8824 32.3012 Skipping because completely internal.
POINT: 30.6787 8.55076 Needs retraction.
POINT: 30.3002 28.8031 Skipping because completely internal.
POINT: 23.4734 33.0583 Skipping because completely internal.
POINT: 24.692 34.2619 Skipping due to short length.
POINT: 25.265 32.1238 Skipping due to short length.
POINT: 26.6407 30.6212 Skipping due to short length.
POINT: 17.8692 30.4739 Skipping because completely internal.
POINT: 13.7227 34.0825 Needs retraction.
POINT: 9.52045 32.3806 Skipping because completely internal.
POINT: 5.65747 32.3806 Skipping because completely internal.
POINT: 32.6045 33.9835 Skipping because completely internal.
POINT: 33.7906 36.5432 Skipping because completely internal.
POINT: 23.6883 38.0078 Needs retraction.
POINT: 3.80012 36.7413 Needs retraction.
POINT: 30.6892 28.9924 Needs retraction.
POINT: 30.9102 1.48282 Skipping because completely internal.
POINT: 30.4394 1.23219 Skipping because completely internal.
POINT: 30.0109 1.45 Skipping due to short length.
POINT: 7.1716 1.8833 Skipping due to short length.
POINT: 4.6331 28.8745 Needs retraction.
POINT: 4.63309 43.8852 Needs retraction.
Moving to layer 23
POINT: 33.2078 43.8852 Needs retraction.
POINT: 33.2078 28.8745 Needs retraction.
POINT: 29.9216 28.4865 Needs retraction.
POINT: 24.8824 32.3012 Skipping because completely internal.
POINT: 30.6787 8.55076 Needs retraction.
POINT: 30.3002 28.8651 Skipping because completely internal.
POINT: 23.4734 33.0583 Skipping because completely internal.
POINT: 24.692 34.2619 Skipping due to short length.
POINT: 25.265 32.1238 Skipping due to short length.
POINT: 26.6407 30.6212 Skipping due to short length.
POINT: 17.8692 30.4739 Skipping because completely internal.
POINT: 13.7227 34.0825 Needs retraction.
POINT: 9.52045 32.3806 Skipping because completely internal.
POINT: 5.65747 32.3806 Skipping because completely internal.
POINT: 32.6045 33.9835 Skipping because completely internal.
POINT: 33.7906 36.5432 Skipping because completely internal.
POINT: 23.6883 38.0078 Needs retraction.
POINT: 3.80012 36.7413 Needs retraction.
POINT: 30.689 29.0543 Needs retraction.
POINT: 30.91 7.75112 Skipping due to short length.
POINT: 30.4394 1.23219 Skipping because completely internal.
POINT: 30.0109 1.45 Skipping due to short length.
POINT: 7.1716 1.8833 Skipping due to short length.
POINT: 4.6331 28.8745 Needs retraction.
POINT: 4.63309 43.8852 Needs retraction.
Moving to layer 24
POINT: 33.2078 43.8852 Needs retraction.
POINT: 33.2078 28.8745 Needs retraction.
POINT: 29.9216 28.5485 Needs retraction.
POINT: 24.8824 32.3012 Skipping because completely internal.
POINT: 30.6787 8.55076 Needs retraction.
POINT: 30.3002 28.927 Skipping because completely internal.
POINT: 23.4734 33.0583 Skipping because completely internal.
POINT: 24.692 34.2619 Skipping due to short length.
POINT: 25.265 32.1238 Skipping due to short length.
POINT: 26.6407 30.6212 Skipping due to short length.
POINT: 17.8692 30.4739 Skipping because completely internal.
POINT: 13.7227 34.0825 Needs retraction.
POINT: 9.52045 32.3806 Skipping because completely internal.
POINT: 5.65747 32.3806 Skipping because completely internal.
POINT: 32.6045 33.9835 Skipping because completely internal.
POINT: 33.7906 36.5432 Skipping because completely internal.
[... 286 lines cut ...]
POINT: 17.8692 30.4739 Skipping because completely internal.
POINT: 13.7227 34.0825 Needs retraction.
POINT: 9.52045 32.3806 Skipping because completely internal.
POINT: 5.65747 32.3806 Skipping because completely internal.
POINT: 32.6045 33.9835 Skipping because completely internal.
POINT: 33.7906 36.5432 Skipping because completely internal.
POINT: 23.6883 38.0078 Needs retraction.
POINT: 3.80012 36.7413 Needs retraction.
POINT: 30.6885 29.1782 Needs retraction.
POINT: 30.91 7.75112 Skipping due to short length.
POINT: 30.4394 1.23219 Skipping because completely internal.
POINT: 30.0109 1.45 Skipping due to short length.
POINT: 7.1716 1.8833 Skipping due to short length.
POINT: 4.6331 28.8745 Needs retraction.
POINT: 4.63309 43.8852 Needs retraction.
````

Of the 25 points identified for travel:
- 10 were identified as needing retraction
- 9 were identified by Slic3r as starting and ending in internal fill (so we don't care)
- 6 were identified by Slic3r to fall under your 2mm travel length.

I can't tell you which travel moves you are looking at here, and gcode.ws doesn't give the directed travel.

[image]

lordofhyphens, 2018-04-10:
For grins, I set the minimum distance before retract to 0 and all of the internal points that were passed over before because of "too short" were inside infill.

VanessaE, 2018-04-10:
(note: I am currently on master branch, at commit ab89469494f6434c6c74f707e6e5065ad8e97159)

Please try the attached files.  cubes.zip contains a simple 20 mm cube, and a 15 mm hollow cubical frame.

Place the cube in the middle of the plate, a bit back from the center, and place the frame to the right of it, a bit forward.  Scale the frame "to size", 20 mm.  Make one copy of each (right click, increase copies).  Put the copied cube to the right of the original, and put the copied frame way over to the left.

Load the attached config (it is not the same file as in the original post) and use it to slice the layout.

The attached gcode file might be useful for comparison - it's the output from just such a layout, as created by the attached config.

When the height gets to where the uprights of the frames are being printed, the print order/pattern should be as in this image ("S" is where the layer starts... well, where the first print moves are anyway, and "E" is the end of it, natch):

[image]

(firmware retract is enabled in that gcode file, so gcode.ws is useless to visualize it, hence my Picasso-quality art, professionally composited onto this Pronterface screenshot :smile: )

All of that prints as expected, except that the retracts for travels 2 and 6 being omitted.

Here's a snippet of the attached gcode from layer 75, which is about where the printer was in the above screenshot, showing from the end of travel move 1, to the start of move 3.  As you can see, there's an unretract (`G11`) when it lands on the front-right upright bit, and a retract (`G10`) as it leaves the front-left bit, and no retraction for travel move no. 2.

```gcode
G11
G92 E0
G1 F10800
G1 X155.760 Y83.566 E0.04928
G1 X155.760 Y81.399 E0.13452
G1 X157.926 Y81.399 E0.21977
G1 X157.926 Y83.566 E0.30501
G1 X157.072 Y83.566 E0.33862
G1 X156.695 Y83.322 F12000.000
G1 X157.072 Y83.109 F12000.000
G1 F10800
G1 X156.217 Y83.109 E0.37227
G1 X156.217 Y81.856 E0.42155
G1 X157.469 Y81.856 E0.47083
G1 X157.469 Y83.109 E0.52011
G1 X157.132 Y83.109 E0.53337
G1 X157.012 Y82.402 F12000.000
G1 F10800
G1 X157.012 Y82.652 E0.54321
G1 X156.674 Y82.652 E0.55653
G1 X156.674 Y82.313 E0.56984
G1 X157.012 Y82.313 E0.58315
G1 X157.012 Y82.342 E0.58426
G1 X140.593 Y82.342 F12000.000 ; <---- retract move is missing!
G1 F10800
G1 X140.593 Y83.566 E0.63243
G1 X138.426 Y83.566 E0.71767
G1 X138.426 Y81.399 E0.80292
G1 X140.593 Y81.399 E0.88816
G1 X140.593 Y82.282 E0.92288
G1 X140.348 Y82.657 F12000.000
G1 X140.136 Y82.282 F12000.000
G1 F10800
G1 X140.136 Y83.109 E0.95542
G1 X138.883 Y83.109 E1.00470
G1 X138.883 Y81.856 E1.05398
G1 X140.136 Y81.856 E1.10326
G1 X140.136 Y82.222 E1.11763
G1 X139.679 Y82.599 F12000.000
G1 F10800
G1 X139.679 Y82.652 E1.11971
G1 X139.340 Y82.652 E1.13302
G1 X139.340 Y82.313 E1.14633
G1 X139.679 Y82.313 E1.15964
G1 X139.679 Y82.539 E1.16852
G10
```

Any random layer between 2.8 and 17.2 mm will be functionally identical, if not necessarily precisely the same values.

Models:  [cubes.zip](https://github.com/slic3r/Slic3r/files/1892476/cubes.zip)
Config:  [slic3r-config-20180409.ini.zip](https://github.com/slic3r/Slic3r/files/1892536/slic3r-config-20180409.ini.zip)
Gcode result:  [Another_Hollow_Cube.gcode.zip](https://github.com/slic3r/Slic3r/files/1892435/Another_Hollow_Cube.gcode.zip)

In case you're curious, I arranged the models the way I did, so that I could pause the print from time to time while the extruder/nozzle is working on that left frame, leaving room to reach in with a sharpie and make reference marks on the cube and frame at the far right.

lordofhyphens, 2018-04-10:
More snippets :)
````
Moving to layer 75
POINT: 0.25 0.25 Needs retraction.
POINT: 0.707079 0.707079 Skipping due to short length.
POINT: 15.8138 1.16416 Skipping because completely internal.
POINT: 18.4111 5.48295 Skipping because completely internal.
POINT: 0.25 18.4581 Needs retraction.
POINT: 0.707079 18.5181 Skipping due to short length.
POINT: 14.2548 1.16416 Skipping because completely internal.
POINT: 18.4111 5.48295 Skipping because completely internal.
````
And here's the gcode for layer 75:
````
G1 Z7.600 F12000.000 ; move to next layer (75)
; "after layer change" code goes here
; end of "after layer change" code
G1 X87.410 Y119.091 F12000.000 ; move to first perimeter point
G11 ; unretract extruder 0
G92 E0 ; reset extrusion distance
G1 F7200
G1 X106.910 Y119.091 E0.76720 ; perimeter
G1 X106.910 Y138.591 E1.53440 ; perimeter
G1 X87.410 Y138.591 E2.30161 ; perimeter
G1 X87.410 Y119.151 E3.06645 ; perimeter
G1 X87.410 Y119.091 F12000.000 ; move inwards before travel
G1 X87.867 Y119.548 F12000.000 ; move to first perimeter point
G1 F14400
G1 X106.453 Y119.548 E3.79768 ; perimeter
G1 X106.453 Y138.134 E4.52892 ; perimeter
G1 X87.867 Y138.134 E5.26015 ; perimeter
G1 X87.867 Y119.608 E5.98903 ; perimeter
G1 X102.974 Y120.005 F12000.000 ; move to first perimeter point
G1 F14400
G1 X105.996 Y120.005 E6.10793 ; perimeter
G1 X105.996 Y137.677 E6.80320 ; perimeter
G1 X88.324 Y137.677 E7.49847 ; perimeter
G1 X88.324 Y120.005 E8.19374 ; perimeter
G1 X102.914 Y120.005 E8.76775 ; perimeter
G1 X105.571 Y124.324 F12000.000 ; move to first infill point
G1 F14400
G1 X101.677 Y120.430 E8.98441 ; infill
G1 X95.213 Y120.430 E9.23873 ; infill
G1 X105.571 Y130.788 E9.81506 ; infill
G1 X105.571 Y137.252 E10.06938 ; infill
G1 X88.749 Y120.430 E11.00538 ; infill
G1 X88.749 Y126.894 E11.25970 ; infill
G1 X99.107 Y137.252 E11.83603 ; infill
G1 X92.643 Y137.252 E12.09035 ; infill
G1 X88.749 Y133.358 E12.30701 ; infill
G10 ; retract extruder 0
G92 E0 ; reset extrusion distance
G1 X138.491 Y133.358 F12000.000 ; move to first perimeter point
G11 ; unretract extruder 0
G92 E0 ; reset extrusion distance
G1 F7200
G1 X138.491 Y115.150 E0.71637 ; perimeter
G1 X157.991 Y115.150 E1.48357 ; perimeter
G1 X157.991 Y134.650 E2.25078 ; perimeter
G1 X138.491 Y134.650 E3.01798 ; perimeter
G1 X138.491 Y133.418 E3.06645 ; perimeter
G1 X138.699 Y133.016 F12000.000 ; move inwards before travel
G1 X138.948 Y133.418 F12000.000 ; move to first perimeter point
G1 F14400
G1 X138.948 Y115.607 E3.76720 ; perimeter
G1 X157.534 Y115.607 E4.49843 ; perimeter
G1 X157.534 Y134.193 E5.22967 ; perimeter
G1 X138.948 Y134.193 E5.96090 ; perimeter
G1 X138.948 Y133.478 E5.98903 ; perimeter
G1 X152.496 Y116.064 F12000.000 ; move to first perimeter point
G1 F14400
G1 X157.077 Y116.064 E6.16926 ; perimeter
G1 X157.077 Y133.736 E6.86453 ; perimeter
G1 X139.405 Y133.736 E7.55980 ; perimeter
G1 X139.405 Y116.064 E8.25507 ; perimeter
G1 X152.436 Y116.064 E8.76775 ; perimeter
G1 X156.652 Y120.383 F12000.000 ; move to first infill point
G1 F14400
G1 X152.758 Y116.489 E8.98441 ; infill
G1 X146.294 Y116.489 E9.23873 ; infill
G1 X156.652 Y126.847 E9.81506 ; infill
G1 X156.652 Y133.311 E10.06938 ; infill
G1 X139.830 Y116.489 E11.00538 ; infill
G1 X139.830 Y122.953 E11.25970 ; infill
G1 X150.188 Y133.311 E11.83603 ; infill
G1 X143.724 Y133.311 E12.09035 ; infill
G1 X139.830 Y129.417 E12.30701 ; infill
; "before layer change" code goes here
; end of "before layer change" code
G10 ; retract extruder 0
G92 E0 ; reset extrusion distance
````

lordofhyphens, 2018-04-10:
[export-gcode.zip](https://github.com/slic3r/Slic3r/files/1892724/export-gcode.zip)

VanessaE, 2019-03-24:
Another example, tested with 05a2f9028a35c56e039cf7a531d63b7ff7e97645 (plus supermerill's thin wall PR):

[image]

In this image, the left hole is not invoking retract for the travel moves from one side to the other.  The hole to the right, on the other hand, is getting retractions properly.

The dashed curved arrow is the travel from elsewhere to the area of interest.  Straight arrows are the print moves.  Curved, solid arrow is the travel move across the left hole after completing the perimeter loop to its right.

This is part of a slightly larger plate which I made to test some problem parts.

Those parts, plated and exported to 3mf, with config:  [retraction.zip](https://github.com/slic3r/Slic3r/files/2999956/retraction.zip)