slic3r/Slic3r #4193: Build errors (XS test failures) under Linux Mint 17.3

anoved, 2017-11-19:
### Version

Build failure encountered while attempting to build master branch (had not previously updated my copy since late 2016). Using `git bisect`, I determined I cannot build commit e064fdf17017fc4a51e7e029dc2a8464a4e6d6f2 or later. I can build and run earlier commits. 

### Operating system type + version

```
LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch:security-4.0-amd64:security-4.0-noarch:security-4.1-amd64:security-4.1-noarch
Distributor ID:	LinuxMint
Description:	Linux Mint 17.3 Rosa
Release:	17.3
Codename:	rosa
```

### Behavior

* _Describe the problem_

Failure while attempting to [build Slic3r](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux#build-slic3r) from master branch at commit e064fdf17017fc4a51e7e029dc2a8464a4e6d6f2 or later.

* _Steps needed to reproduce the problem_

Check out a recent commit (version noted above or later) and attempt `perl Build.PL`.

* _Expected Results_

Build succeeds; I can proceed to build with `--gui` option and run Slic3r.

* _Actual Results_

Build fails with XS error output reported below. 

My assumption is something is mangled in my system's Perl installation but I don't know how to proceed with diagnosis from here. 

#### perl Build.PL output:

```
App::cpanminus is up to date. (1.7043)
Class::XSAccessor is up to date. (1.19)
Devel::CheckLib is up to date. (1.11)
Encode is up to date. (2.93)
Encode::Locale is up to date. (1.05)
ExtUtils::CppGuess is up to date. (0.12)
ExtUtils::MakeMaker is up to date. (7.30)
ExtUtils::ParseXS is up to date. (3.35)
skipping S/SH/SHAY/perl-5.26.1.tar.bz2
File::Spec is up to date. (3.62)
Getopt::Long is up to date. (2.5)
IO::Scalar is up to date. (2.111)
Module::Build::WithXSpp is up to date. (0.14)
Moo is up to date. (2.003003)
skipping S/SH/SHAY/perl-5.26.1.tar.bz2
Scalar::Util is up to date. (1.49)
Test::Harness is up to date. (3.39)
Test::More is up to date. (1.302111)
Thread::Queue is up to date. (3.12)
Thread::Semaphore is up to date. (2.13)
Time::HiRes is up to date. (1.9746)
Unicode::Normalize is up to date. (1.25)
threads is up to date. (2.16)
threads::shared is up to date. (1.57)
Cleaning up build files
Cleaning up configuration files
Not in MANIFEST: src/libslic3r/utils.hpp
MANIFEST appears to be out of sync with the distribution
cpanm (App::cpanminus) 1.7043 on perl 5.018002 built for x86_64-linux-gnu-thread-multi
Work directory is /home/anoved/.cpanm/work/1511110801.7468
You have make /usr/bin/make
You have LWP 6.29
You have /bin/tar: tar (GNU tar) 1.27.1
[... 477 lines cut ...]
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by John Gilmore and Jay Fenlason.
You have /usr/bin/unzip
--> Working on ./xs
Entering /home/anoved/Repos/Slic3r-dev/xs
Checking if you have Module::Build 0.38 ... Yes (0.4204)
Checking if you have ExtUtils::Install 1.46 ... Yes (2.06)
Configuring /home/anoved/Repos/Slic3r-dev/xs
Running Build.PL
-> OK
Checking dependencies from MYMETA.json ...
Checking if you have ExtUtils::ParseXS 3.18 ... Yes (3.35)
Checking if you have ExtUtils::Typemaps::Default 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps::Basic 1.05 ... Yes (1.05)
Checking if you have ExtUtils::Typemaps 1.00 ... Yes (3.35)
Checking if you have ExtUtils::XSpp 0.18 ... Yes (0.18)
Checking if you have Module::Build 0.3601 ... Yes (0.4204)
Checking if you have Test::More 0 ... Yes (1.302111)
Building and testing Slic3r-XS-0.01
-> FAIL Installing ./xs failed. See /home/anoved/.cpanm/work/1511110801.7468/build.log for details. Retry with --force to force install it.
```

lordofhyphens, 2017-11-19:
Probably related to local::lib.

Since the build server is running tests fine...

On Nov 19, 2017 11:39 AM, "Jim" <notifications@github.com> wrote:

> Version
>
> Build failure encountered while attempting to build master branch (had not
> previously updated my copy since late 2016). Using git bisect, I
> determined I cannot build commit e064fdf
> <https://github.com/alexrj/Slic3r/commit/e064fdf17017fc4a51e7e029dc2a8464a4e6d6f2>
> or later. I can build and run earlier commits.
> Operating system type + version
>
> LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch:security-4.0-amd64:security-4.0-noarch:security-4.1-amd64:security-4.1-noarch
> Distributor ID:	LinuxMint
> Description:	Linux Mint 17.3 Rosa
> Release:	17.3
> Codename:	rosa
>
> Behavior
>
>    - *Describe the problem*
>
> Failure while attempting to build Slic3r
> <https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux#build-slic3r>
> from master branch at commit e064fdf
> <https://github.com/alexrj/Slic3r/commit/e064fdf17017fc4a51e7e029dc2a8464a4e6d6f2>
> or later.
>
>    - *Steps needed to reproduce the problem*
>
> Check out a recent commit (version noted above or later) and attempt perl
> Build.PL.
>
>    - *Expected Results*
>
> Build succeeds; I can proceed to build with --gui option and run Slic3r.
>
>    - *Actual Results*
>
> Build fails with XS error output reported below.
>
> My assumption is something is mangled in my system's Perl installation but
> I don't know how to proceed with diagnosis from here.
> perl Build.PL output:
>
> App::cpanminus is up to date. (1.7043)
> Class::XSAccessor is up to date. (1.19)
> Devel::CheckLib is up to date. (1.11)
> Encode is up to date. (2.93)
> Encode::Locale is up to date. (1.05)
> ExtUtils::CppGuess is up to date. (0.12)
> ExtUtils::MakeMaker is up to date. (7.30)
> ExtUtils::ParseXS is up to date. (3.35)
> skipping S/SH/SHAY/perl-5.26.1.tar.bz2
> File::Spec is up to date. (3.62)
> Getopt::Long is up to date. (2.5)
> IO::Scalar is up to date. (2.111)
> Module::Build::WithXSpp is up to date. (0.14)
> Moo is up to date. (2.003003)
> skipping S/SH/SHAY/perl-5.26.1.tar.bz2
> Scalar::Util is up to date. (1.49)
> Test::Harness is up to date. (3.39)
> Test::More is up to date. (1.302111)
> Thread::Queue is up to date. (3.12)
> Thread::Semaphore is up to date. (2.13)
> Time::HiRes is up to date. (1.9746)
> Unicode::Normalize is up to date. (1.25)
[... 292 lines cut ...]
> t/04_expolygon.t ................ Can't load '/home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZNSt20regex_token_iteratorIN9__gnu_cxx17__normal_iteratorIPKcSsEEcSt12regex_traitsIcEEneERKS7_ at /usr/lib/perl/5.18/DynaLoader.pm line 184.
>  at t/04_expolygon.t line 7.
> Compilation failed in require at t/04_expolygon.t line 7.
> BEGIN failed--compilation aborted at t/04_expolygon.t line 7.
> t/04_expolygon.t ................ Dubious, test returned 2 (wstat 512, 0x200)
> No subtests run
> t/05_surface.t .................. Can't load '/home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZNSt20regex_token_iteratorIN9__gnu_cxx17__normal_iteratorIPKcSsEEcSt12regex_traitsIcEEneERKS7_ at /usr/lib/perl/5.18/DynaLoader.pm line 184.
>  at t/05_surface.t line 6.
> Compilation failed in require at t/05_surface.t line 6.
> BEGIN failed--compilation aborted at t/05_surface.t line 6.
> t/05_surface.t .................. Dubious, test returned 2 (wstat 512, 0x200)
> No subtests run
> t/06_polygon.t .................. Can't load '/home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZNSt20regex_token_iteratorIN9__gnu_cxx17__normal_iteratorIPKcSsEEcSt12regex_traitsIcEEneERKS7_ at /usr/lib/perl/5.18/DynaLoader.pm line 184.
>  at t/06_polygon.t line 7.
> Compilation failed in require at t/06_polygon.t line 7.
> BEGIN failed--compilation aborted at t/06_polygon.t line 7.
> t/06_polygon.t .................. Dubious, test returned 2 (wstat 512, 0x200)
> No subtests run
> t/07_extrusionpath.t ............ Can't load '/home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so: undefined symbol: _ZNSt20regex_token_iteratorIN9__gnu_cxx17__normal_iteratorIPKcSsEEcSt12regex_traitsIcEEneERKS7_ at /usr/lib/perl/5.18/DynaLoader.pm line 184.
>  at t/07_extrusionpath.t line 6.
> Compilation failed in require at t/07_extrusionpath.t line 6.
> BEGIN failed--compilation aborted at t/07_extrusionpath.t line 6.
> t/07_extrusionpath.t ............ Dubious, test returned 2 (wstat 512, 0x200)
> No subtests run
> t/08_extrusionloop.t ............ Can't load '/home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /home/anoved/Repos/Slic3r-dev/xs/blib/arch/auto/Slic3r/XS/XS.so: und

lordofhyphens, 2017-11-19:
Ah, wait. Check your compiler. Gcc 4.9 or later (or another c++11 compiler)
is *required* for Slic3r.

On Nov 19, 2017 12:05 PM, "Joe Lenox" <lordofhyphens@gmail.com> wrote:

> Probably related to local::lib.
>
> Since the build server is running tests fine...
>
>
> On Nov 19, 2017 11:39 AM, "Jim" <notifications@github.com> wrote:

anoved, 2017-11-19:
Ah, good catch. That's it! Thanks.

For future reference for anyone trying to build on a system like mine, I [added](https://superuser.com/questions/772954/install-gcc-g-version-4-9-in-linux-mint#796621) the [upstream toolchain](https://wiki.ubuntu.com/ToolChain#PPA_packages) to get gcc-4.9 and [updated my cc/c++ symlinks](https://askubuntu.com/questions/26498/choose-gcc-and-g-version#26518) to point to the new version. Now the build succeeds.

closed 2017-11-19 by anoved

lordofhyphens, 2017-11-20:
The build system respects environment variables CXX and CC.

iznogoud06, 2018-03-28:
You made my day @lordofhyphens ! Many thanks !!!

For centos users: gcc 4.9 is available here:

yum install centos-release-scl-rh          
yum install devtoolset-3-gcc devtoolset-3-gcc-c++
and update PATH in ~/.bashrc : PATH="/opt/rh/devtoolset-3/root/usr/bin/:$PATH";