slic3r/Slic3r #4195: New speed feature.

dobrzann, 2017-11-22:
### Version
1.3.0

### Operating system type + version
Win 10 x64

### Behavior
last time I was tried to print PLA model with PET-G support structure but problem is PLA not sticking to PET-G well, but when I tried to decreasing speed and turn cooling off during printing first layer of model after last layer of support interface It was look much better.

So I think the new speed option will be very usefull, for example "Speed of first N layers after support" and "turn off cooling when printing first N layers after support

I mean this layer.
[image]

And I wanted to ask is it possible to add Skirt for extruder 1, as it is in extruder 0.  It will be usefull too when printing something where the model perimeters and support material structure are on the firts layer.

_Is this a new feature request?_
YES

lordofhyphens, 2017-11-22:
One feature request at a time, please.

dobrzann, 2017-11-22:
Sorry, already corrected