slic3r/Slic3r #4197: Modify GUI --> add new tab

larajustine, 2017-11-24:
Hi, I'm trying to add a new option in the settings in order to modify pressure parameters, but I didn't find out which perl file contains this information.

As a last resort, I could modify just the GCode, but I also don't know which perl file I would use.

Any help? :)

lordofhyphens, 2017-11-28:
Sorry it's taken me this long to get to you. 
I have a wiki entry for this. A good heuristic to find everything is to check for an already-existing option and see where it is defined. Feel free to add notes to the wiki.

https://github.com/alexrj/Slic3r/wiki/Code%3A-Adding-new-option-to-GUI-CLI

closed 2017-11-28 by lordofhyphens

larajustine, 2017-11-29:
How can I do this check? 
I think just running `perl slic3r.pl` is not actualizing the .o files.

lordofhyphens, 2017-11-30:
You have to run Build.pl to install the compiled library.

On the Perl side it is immediate upon restart of Slic3r

larajustine, 2017-12-01:
Even doing this, the same error occurs. 
Do you think this is because the Wx error? When I run `perl build.pl --gui` I get this error:
`Installing Wx failed. See C:\Users\laraj\.cpanm\work\1512136829.5664\build.log for details. Retry with --force to force install it.`
`Don't worry, this module is optional.`

But it says it's optional :(

lordofhyphens, 2017-12-01:
It is not optional for the GUI. It is optional for the CLI.

I suggest that if you are on Windows, use Slic3r perl and the prebuilt
local-lib archive I produced for it.

larajustine, 2017-12-04:
Mute the thread? What do you mean?

lordofhyphens, 2017-12-04:
Mute the thread is a message from GitHub (I sometimes reply via email). Ignore it.

lordofhyphens, 2017-12-04:
Regarding your error, sounds like you broke something when editing the XS side of things.