slic3r/Slic3r #4201: Under-extrusion after long bridges

VanessaE, 2017-11-26:
### Version

commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior

After laying a long enough bridge, Slic3r is dropping the flow rate of the rest of the layer to around half the correct amount, as observed by watching the speed of my extruder's gears and testing by bumping the flow to 200% in Pronterface while it's still going. This is after 20+ absolutely flawless layers.  I did not let the print continue after verifying the bumping the flow rate up worked, so I can't say when the starvation ends.

I can pull back on the incoming filament and feel the extruder drawing it in solidly.

I have otherwise verified nine way from Sunday that the extruder is performing flawlessly and that there are no restrictions in the filament path, that the hot end is no clogged (did a few cold pulls to check), and that there are no mechanical or electronic problems.

I am NOT using pressure advance or LIN_ADVANCE.

I was printing at 30 mm/sec, 0.1 mm layers, 0.4 mm line width.

Using larger line widths for some features (e.g. first layer) may also trigger this issue.

I am unable to print because of this. This is a hard blocker.

#### STL/Config (.ZIP) where problem occurs

config: [config-20171126.ini.zip](https://github.com/alexrj/Slic3r/files/1504172/config-20171126.ini.zip)
stl: [J-head v8 extruder MK2 - all plated no gears.stl.zip](https://github.com/alexrj/Slic3r/files/1504174/J-head.v8.extruder.MK2.-.all.plated.no.gears.stl.zip)

lordofhyphens, 2017-11-26:
I suspect that the pressure is able to go to near 0 after a particular long
bridge span and Slic3r doesn't have a good way to model this in such a way
as it can effectively over extrude to build pressure back up (similar to
the priming move most people use a skirt for).

Is the problem lessened if you bridge faster? Is there anything in the
gcode itself that indicates that Slic3r is deliberately under extruding
(from a flow perspective)? That distinction changes the behavior from a bug
(aka slic3r is doing something wrong from what it knows) to something that
requires more study  (Slic3r is missing something in this case, but the
solution is nonobvious).

VanessaE, 2017-11-26:
I don;t understand wtf is happening.  The gcode looks right - the same batch of line gets the same extrude distance, from one layer before the problem to one layer after.  Still, if it were a pressure problem, wouldn't it look more like *over*extrusion? These are 0.1 mm layers, so a bridge would need several times the layer's normal flow rate, and the problem seems to start with the first few sparse infill lines printed right after the bridge.

VanessaE, 2017-11-26:
Here's the gcode output, maybe you can see the problem?

[J-head v8 extruder MK2 - all plated no gears.gcode.zip](https://github.com/alexrj/Slic3r/files/1504278/J-head.v8.extruder.MK2.-.all.plated.no.gears.gcode.zip)

(8.7 MB zip, expands to 29 MB)

VanessaE, 2017-11-26:
*looks at gcode again*

Well, I *thought* I had pressure advance disabled.  Still the setting is so low it ought to have almost no effect on a bowden? (my backup setup, my primary extruder broke)

VanessaE, 2017-11-27:
Figured it out.  Xcarriage gets pushed up by e.g a curled/warped overhang, and I guess the Z nuts get suck that way and don't recover until sometime later. Guess it's time to make some Z isolators and stiffen up my preloaders.

closed 2017-11-27 by VanessaE

lordofhyphens, 2017-11-27:
Well, I'm glad that you were able to diagnose what was going on (and extra glad it didn't result in more work on Slic3r :) )