slic3r/Slic3r #4202: Large gaps between extrusions while using large diameter nozzles

zappyguy, 2017-11-28:
### Version 
_1.3.0-dev_

### Operating system type + version
_Windows 10_

### Behavior
When printing with a nozzle diameter of 0.8mm the printer will leave large gaps between everything. The distance from centre to centre is about 1~1.25mm.
Extrusion widths are all set to either default or automatic.
Extrusion multiplier is set correctly.
Slower print speeds have been tested to see if thermal restriction is the cause - it is not.
Layer heights are accurate.

#### STL/Config (.ZIP) where problem occurs
I recall this being an issue while using 1.2.9, I cannot specify where the error is occurring, but it is consistent

Photo:
[image]
.

lordofhyphens, 2017-11-28:
You can see what Slic3r thinks is the extrusion width at the exported gcode.
A layer screenshot from Slic3r's own preview is also important. 

You are also missing a test model and an exported config (Ctrl+E).

bubnikv, 2017-11-28:
It is physically impossible to extrude a wide extrusion for most of the
plastics through a thin nozzle. The plastic viscosity is simply too high
for that.

zappyguy, 2017-11-28:
Hey Lordofhyphens
Print model:
https://www.thingiverse.com/thing:648772
I checked the gcode and it did specify the nozzle diameter to be 0.8mm.
Slic3r's preview (not layer view) shows the parts to be correctly filled.
-Screenshot:
[image]
-Config:
[800µm_oak_config.txt](https://github.com/alexrj/Slic3r/files/1510827/800.m_oak_config.txt)

Post post update:
After seeing Slic3r's preview image I decided to run a second print at 50% speed (because it's 3am and I don't want to wake my housemates) and got these results:
https://imgur.com/n1Dzgic.jpg
There are some holes here and there, but it's much better than what came out before. It seems that most of this was caused by the E3D-V6 struggling to keep the heat up for an 800µm nozzle. I've since slowed it down to 35% speed.
I did review the original print and found that the extrusions were exactly 800µm, which I think they are not supposed to be. I suspect the automatic extrusion width calculator (?) tries to up the extrusion width to ensure good layer adhesion. 
I'll post some more results once it finishes up later this morning.

lordofhyphens, 2017-11-28:
The auto width performs suboptimally outside of a relatively common combination of nozzle diameters and layer heights, as @bubnikv has noted. The physics and flow aren't very nice.

zappyguy, 2017-12-06:
The problem has been resolved, it is an error on my part, I was extruding more material than my hot-end could manage.

closed 2017-12-06 by lordofhyphens