slic3r/Slic3r #4204: Default view for rectangular plate should have plate centered in the viewport, with rotation axis centered on it, not on its origin (e.g corner)

VanessaE, 2017-11-28:
### Version
commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior
 
Title says it all.

lordofhyphens, 2017-11-28:
Screenshot of how it is now would be handy though.

VanessaE, 2017-11-28:
That is impossible, because the safe point would naturally vary from one print to another.  The idea here is to merge any startup ooze directly into some part of the print that is either discarded or effectively hidden instead of spreading it across the plate.

VanessaE, 2017-11-28:
[image]

lordofhyphens, 2017-11-28:
@VanessaE probably want to move your "that is impossible" to the correct issue.

lordofhyphens, 2017-11-28:
@VanessaE could you set your origin in Printer settings to another point to see the difference?

VanessaE, 2017-11-28:
RE: "that is impossible"  ...um, oops? :smile:

I forgot to mention, the bed shape is from an stl.  Changing the model's origin in Blender doesn't affect it, but moving the model on Blender's grid and exporting to stl moves where Slic3r thinks the plate's origin is, so if I center the model on Blender's grid, Slic3r puts the plate origin at the center of the plate, even if the model's origin were somewhere else on the model.  Slic3r's default view shifts accordingly, and parts are placed near the new origin. Obviously the model's origin needs to be taken into account, at least as far as where the plate's origin is.

If I keep it like that (that is, make Slic3r think the origin is at the center), the exported model has corresponding X/Y offsets. Something placed at the apparent origin shows up there in Slic3r's Preview tab when sliced, but ends up in the corner of the plate when loaded into Pronterface.

Besides, it's still a square plate, so the actual plate origin ought not be in the center.  Just the default view and placement location of loaded parts should be.

Also, I guess the default zoom level should make the plate fit the viewport.

Here's what I see at a fresh start, after loading a small part.  In this view, the bed stl has its model origin at the center of the model's bottom side, and the "corner" of the model in at Blender's grid origin (that model is attached below):
[image]

And here, fresh start, after loading the same small part, with the bed model still having its origin at the model's bottom center, but with the model is positioned with its center at Blender's grid origin before exporting:
[image]

Bed model, with model origin at model's bottom center, corner placed at Blender's grid origin (includes .blend file):
[prusa-i3-bed.zip](https://github.com/alexrj/Slic3r/files/1511715/prusa-i3-bed.zip)