slic3r/Slic3r #4205: Nozzle should move to a good start point before heating

VanessaE, 2017-11-28:
### Version
commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior
At the start of the print, the nozzle is left where it can drool as it heats up.  For some printers, that can be quit some distance above the bed, too.  So when the print actually starts, you're left with a stay line that will mar the print if it ins't plucked off while printing (which may be difficult on some printers.

Instead, the nozxzle should be moved to the start point of the skirt, if any, or the brim, if any (assuming it always spirals inward).  If neither is enabled, put the nozzle somewhere that will have sparse infill immediately after the bottom solid layers are to be printed.  If that cant be found, put the nozzle at X/Y home position.  In all case, put the nozzle all the way down on the bed(Z=0 mm) to sort of "cap" it.

THEN start heating it up,.

### Model/config

None needed.

lordofhyphens, 2017-11-28:
That's the job of start gcode and the user to figure it out where is "safe".

VanessaE, 2017-11-28:
That is impossible, because the safe point would naturally vary from one print to another. The idea here is to merge any startup ooze directly into some part of the print that is either discarded or effectively hidden instead of spreading it across the plate.

VanessaE, 2017-11-28:
Oops didn't mean to close.

lordofhyphens, 2017-11-28:
What you are describing would break custom gcode that needed to touch
heating (if you wanted full automatic).

Perhaps some way of calculating a reference to some point that matters?

VanessaE, 2017-11-28:
Maybe the coords for the first printed line's start point can be made available to the custom gcode?

Itox001, 2017-11-29:
I've found that it's much better to remove the oozing before the print starts, else there's a chance that it will stick to the print head and start blobbing when it actually starts printing. For this, I set my home 3 mm off my bed, so that when the nozzle enters the bed to start printing, it will wipe off excess material on the bed's side. Perhaps you could do something like this??

Tinchus2009, 2017-12-11:
That is the worokr of start gcode. And the safe area, should be away of onject/brim. Why? because while heating up, some leaking from the filament will happend sometimes. Tha culd mae a blob at the begining of the brim for example, makeing the brim to not atach to bed in the first centimeters. not good

VanessaE, 2019-08-17:
This is no longer needed -- can be addressed with start gcode.

closed 2019-08-17 by VanessaE