slic3r/Slic3r #4207: Slic3r creates extremely long prints if left open too long and settings changed that should have little effect

VanessaE, 2017-11-28:
### Version

commit 202a90ff903be5b61fca317c6b614af0d5f34d8a

### Operating system type + version

Debian 9

### Behavior

* Load the attached model, slice/export to gcode,
* Load the results into Pronterface (I suppose any host software will do)
* Note the estimated print time.
* Leave Slic3r and Pronterface open and just wander off or otherwise ignore them for a few hours.
* Come back to Slic3r and immediately change some innocuous setting in Slic3r that should have no appreciable effect on the print time - maybe turn your retraction speed up or something.  SAVE the altered profile.
* Immediatelt slice and export the already loaded model.
* Load that new result into Promterface
* Note the estimated time - it is now way longer than before, maybe by several hours, even though it perhaps should have been shorter.
* Quit Slic3r and restart.
* Load the model and immediately slice/export without changing you setting(s) back to normal.
* Load that result into Pronterface,
* Note the estimated time.  It should now be similar to your first result.

I don't know how many settings or which ones will trigger this, and I didn't compare the gcode outputs.

Also, you will notice that Slic3r became extremely sluggish after that first slice.  Every action now takes several seconds to complete, even if background processing is disabled (as it is for me).  This should be addressed as well.

#### STL/Config (.ZIP) where problem occurs
config:  [config-20171128.ini.zip](https://github.com/alexrj/Slic3r/files/1511364/config-20171128.ini.zip)
model:  [J-head v8 extruder MK2 - plated - most parts, no gears.stl.zip](https://github.com/alexrj/Slic3r/files/1511365/J-head.v8.extruder.MK2.-.plated.-.most.parts.no.gears.stl.zip)

lordofhyphens, 2017-11-30:
Print temp would be a better thing to use to test; retraction speed would affect your print time

VanessaE, 2017-12-01:
That is why I suggested turning the speed up, so that the second sliced output should theoretically have a faster estimated print time. Didn't think to suggest changing the temperature - that's a better idea for sure.